sap-labs-france/ev-server #2526: Locking: busy loop implementation for lock acquisition with timeout stall almost totally the node.js event loop.

jerome-benoit, 2021-06-07:
Introduced by https://github.com/sap-labs-france/ev-server/pull/2518 

Benchmarked code sketches https://github.com/jerome-benoit/benchmarks-js/blob/master/busy-wait.js: 

```
dummyTimeoutBusyWait x 0.50 ops/sec ±0.01% (6 runs sampled)
divideAndConquerTimeoutBusyWait x 376,241 ops/sec ±14.81% (53 runs sampled)
setIntervalTimeoutBusyWait x 605,100 ops/sec ±14.35% (42 runs sampled)
Fastest is setIntervalTimeoutBusyWait
```

Not even a full one single full operation is allowed to be executed with the current implementation. 

As a side note regarding the handling of that issue: there's no need to be a performance expert to know that with a language running on a mono-threaded interpreter, calling a function in a loop condition is a one the 'never ever do that' code pattern.

LucasBrazi06, 2021-06-07:
This is interesting indeed but I would have expected your code VS my code with the sleep of 1 sec still in the loop + a call to the DB for both implementation and show me the gain between the two.
But I don't need performance improvements where there is no need to.
And if you look at the numerous loops in the project calling functions, this is the least of my concerns to be honest.

jerome-benoit, 2021-06-08:
> This is interesting indeed but I would have expected your code VS my code with the sleep of 1 sec still in the loop + a call to the DB for both implementation and show me the gain between the two.
> 
It’s absolutely not needed, the simple benchmarking of the code pattern give the proof of its utterly inefficiency.
> But I don't need performance improvements where there is no need to.
> And if you look at the numerous loops in the project calling functions, this is the least of my concerns to be honest.
> 

Blocking the event loop totally is the least of your concern, seriously?

I will let you explain that to the customer that will have more than X seconds wait at ... a start transaction? … a stop transaction? …

The way your are handling that issue is not understandable: it’s badly managed (I will discuss it with the management and the human resource that week, some aspects are unacceptable) and show a dysfunctional management.

--
Jérôme Benoit aka fraggle
Piment Noir - https://piment-noir.org
OpenPGP Key ID : 27B535D3
Key fingerprint : B799 BBF6 8EC8 911B B8D7 CDBC C3B1 92C6 27B5 35D3

LucasBrazi06, 2021-06-08:
Again, in the LockingManager class, with a sleep of 1s and an access to the DB in the while loop, the Date.now() will be called once every 1.2s at worse so I don't get what is blocking the event loop in this scenario!?

Moreover, the way you present the problem is not acceptable to me and this is not the first time.
I'm the dev lead, you present me an issue, we discuss it and in the end I take a decision whether you like it or not.

You are always in aggressive communication mode, writing understandable and lengthy emails/comments, not only with me, but also with customers/colleagues who already complained about this behaviour and despite you know that, you still making the same mistakes again, not learning from your mistakes.

You also still continuously provides code without testing it, delivering bugs in QA or production, again not learning from your mistakes.

For me you are creating more problems than you solve, consuming my precious time and generating more stress than the entire project itself.

I talked to your manager and your responsibility will be restrained to few topics you master, not related to me at least.

I cannot work with someone who does not learn from his mistakes.

closed 2021-06-08 by LucasBrazi06

jerome-benoit, 2021-06-08:
> Moreover, the way you present the problem is not acceptable to me and this is not the first time.
> I'm the dev lead, you present me an issue, we discuss it and in the end I take a decision whether [...]

You continue to project your own behaviour on me for most of the aspects you cite and in short you describe here your own dysfunctional vertical management.

--
Jérôme Benoit aka fraggle
Piment Noir - https://piment-noir.org
OpenPGP Key ID : 27B535D3
Key fingerprint : B799 BBF6 8EC8 911B B8D7 CDBC C3B1 92C6 27B5 35D3