sap-labs-france/ev-server #2543: Storage > Use Tenant object instead of tenant id to access storage classes

OliveGerste, 2021-06-11:
Ideally raising one PR per storage class

LucasBrazi06, 2021-06-17:
@RohanAntony Good job, you can continue on the other storages.
It would be nice to have a list of checkboxes (one per storage) in the issue description to track the progression.
Thanks.

RohanAntony, 2021-06-17:
@LucasBrazi06 Thank you

RohanAntony, 2021-06-17:
Required storage class changes

- [ ] AssetStorage
- [X] BillingStorage
- [ ] CarStorage
- [ ] ChargingStationStorage
- [ ] CompanyStorage
- [ ] ConnectionStorage
- [ ] ConsumptionStorage
- [ ] LoggingStorage
- [ ] NotificationStorage
- [ ] OCPIEndpointStorage
- [X] OCPPStorage (Fixed in another issue)
- [ ] OICPEndpointStorage
- [ ] RegistrationTokenStorage
- [ ] SettingStorage
- [ ] SiteAreaStorage
- [ ] SiteStorage
- [ ] StatisticsStorage
- [ ] TagStorage
- [ ] TenantStorage
- [ ] TransactionStorage
- [ ] UserStorage

LucasBrazi06, 2021-07-28:
@RohanAntony Just merged Charging Station's PRs, THANKS!
You can undraft the others when you'll be ready.
Put back to In Progress in the meantime.

LucasBrazi06, 2021-07-28:
@RohanAntony Next. ;-)

closed 2021-09-02 by LucasBrazi06