sap-labs-france/ev-server #2597: User schema validation: ID is missing on a request on QA landscape (testcharger tenant)

jerome-benoit, 2021-06-25:
```
[
      {
            "stack": "Error: Error: should have required property 'ID'
   at UserValidator.validate (/home/vcap/app/dist/webpack:/src/server/rest/v1/validator/SchemaValidator.ts:71:13)
   at UserValidator.validateUserGetByID (/home/vcap/app/dist/webpack:/src/server/rest/v1/validator/UserValidator.ts:52:10)
   at RequestMapper.handleGetUserImage (/home/vcap/app/dist/webpack:/src/server/rest/v1/service/UserService.ts:313:48)
   at CentralRestServer.restServiceSecured (/home/vcap/app/dist/webpack:/src/server/rest/CentralRestServerService.ts:382:13)",
            "previous": {
                  "errors": [
                        {
                              "path": "",
                              "message": "should have required property 'ID'"
                        }
                  ],
                  "content": {},
                  "schema": {
                        "title": "Get a User by id",
                        "type": "object",
                        "definitions": {
                              "ID": {
                                    "$ref": "common.json#/definitions/id"
                              },
                              "ProjectFields": {
                                    "$ref": "common.json#/definitions/projectFields"
                              }
                        },
                        "required": [
                              "ID"
                        ]
                  }
            }
      }
]
```

Ninjeneer, 2021-06-25:
The error has been thrown because an incomplete request has been sent (11:22:28): `/client/api/UserImage` without providing an ID.

By the way, as an invalid request is not an error coming from the server, shouldn't we return a 409 or 422 error, without flagging it as an error in the logs ?

LucasBrazi06, 2021-06-25:
We will refactor the HTTP error codes and think about fixing this kind of issue.
I like when we log it, we have a trace of the issue.

jerome-benoit, 2021-06-25:
> By the way, as an invalid request is not an error coming from the server, shouldn't we return a 400 or 422 error, without flagging it as an error in the logs ?

400. 422 is WebDAV specific.

jerome-benoit, 2021-06-25:
> The error has been thrown because an incomplete request has been sent (11:22:28): `/client/api/UserImage` without providing an ID.

I think you have to found the culprit ;) Otherwise that means we have a bug for that endpoint callee code somewhere.

Ninjeneer, 2021-06-25:
> I think you have to found the culprit ;) Otherwise that means we have a bug for that endpoint callee in the client code somewhere.

The old endpoint is still used in the mobile app, as we first release server+dashboard before migrating mobile routes. However the branch is already ready :)

jerome-benoit, 2021-06-25:
All REST endpoints MUST be backward compatible with all ev-mobile version released so far. It's a showstopper otherwise.

Ninjeneer, 2021-06-25:
Compatibility is still maintained, that's why I redefine every params in routers (e.g : req.query.ID = req.params.id etc...) 
The error here is not compatibility issue, I think Alix just tried/failed a call, as the mobile HTTP call explicitely send an ID

But you're right, in the long term we should implement API versionning

closed 2021-07-31 by LucasBrazi06