sap-labs-france/ev-server #2641: Admin without tag cannot stop any transaction

Ninjeneer, 2021-07-06:
Admin without any tag (or haven't logged off/in after new tag assignment) cannot stop any transaction

Reproduction steps :

- Log as a random user and start a remote transaction
- Log as an admin without any tags and try to stop the transaction

[image]

The tags of the user sending the stop command are checked, while it should not be for admins
[image]

closed 2021-07-31 by LucasBrazi06

hughsheehy, 2022-07-16:
When was this completed? I'm having this error today. 

The logs say 

User API > OCPP Command 'RemoteStopTransaction' has failed: The user does not have any active badge

hughsheehy, 2022-07-16:
I got the pPUT API to work, but I had to add the tagID and visualTagID into the args payload.