sap-labs-france/ev-server #2749: smtp not working

rccess, 2021-08-20:


LucasBrazi06, 2021-08-23:
What is the relation between an error in the front-end console and the backend which is sending the emails.
You should have a look at the logs.
Let me know.

rccess, 2021-08-24:
where i can see specific smtp logs ?

can you please suggest me logs file, since nothing in showing in common npm logs

and no any error showing in console of command line

Please do any hint so i can do further troubleshoot

LucasBrazi06, 2021-08-24:
When I say logs I mean the 'logs' collection in MongoDB.

rccess, 2021-08-24:
smtp only working with 465 , not working with 587 and 25 port

 "subject": "e-Mobility - test - Reset Password\n",
            "error": "Error: connection encountered an error (140315961976704:error:1408F10B:SSL routines:ssl3_get_record:wrong version number:../deps/openssl/openssl/ssl/record/ssl3_record.c:332:\n)

Ninjeneer, 2021-08-24:
> I have defined smtp in config.json of ev-server/src/assets/config.json
> 
> "Email": {
> "smtp": {
> "from": "[care@xxxxxxxxxx.info](mailto:care@xxxxxxxxxx.info)",
> "host": "xxxxxx.xxxxxxxxxx.one",
> "port": 25,
> "secure": true,
> "requireTLS": false,
> "user": "ev",
> "password": "xxxxxxxxx"
> },
> "smtpBackup": {
> "from": "[care@xxxxxxxxxx.info](mailto:care@xxxxxxxxxx.info)",
> "host": "xxxxxx.xxxxxxxxxx.one",
> "port": 25,
> "secure": true,
> "requireTLS": false,
> "user": "ev",
> "password": "xxxxxxxxx"
> }
> },
> 
> but signup and password reset emails are not going , smtp is working fine in other portal
> Note: there is not firewall or sender id restrictions
> 
> console error is
> 
> [image]

Hi,

Please, avoid using the port 25. in addition to not being secured, most of the Internet Service Providers and Cloud providers are blocking this port, which can lead to issues. Consider using only port 465 or 587.

> smtp only working with 465 , not working with 587 and 25 port
> 
> "subject": "e-Mobility - test - Reset Password\n",
> "error": "Error: connection encountered an error (140315961976704:error:1408F10B:SSL routines:ssl3_get_record:wrong version number:../deps/openssl/openssl/ssl/record/ssl3_record.c:332:\n)
> 
> "Email": {
> "smtp": {
> "from": "[care@vxxxx.in](mailto:care@vxxxx.in)",
> "host": "smtp.zexxxxxxxil.com",
> "port": 587,
> "secure": true,
> "requireTLS": false,
> "user": "[care@xxxxxx.in](mailto:care@xxxxxx.in)",
> "password": "wSsVR61/8xL0xxxxxf07SSqSqrKpccyl0HIUxxxxx7E2QWxxxxzQlSDSiF9mqxxxxxxnvhDzxxxxxOILI4Jzg1uxxxEMoi+g==",
> "debug": true
> },
> 
> Note: tried with "requireTLS": false, and true , but same issue issue

Is your mail server well configured for SSL **and** for TLS ? Could you please try with `port: 465`, `secure: true` and `requireTLS: false` ?

LucasBrazi06, 2021-09-02:
Closed for inactivity reason.

closed 2021-09-02 by LucasBrazi06