sap-labs-france/ev-server #2831: Docker setup point to a nonexistent url

niahane, 2021-09-20:
Hi, i am trying to build a docker setup of this server, i follow the guide but at certain point it ask me for username and password of github.
Now github seems to not working with user and pass anymore so i have generated a personal token and in the install wizard i inserted the token witout password but in next step it freeze because:
remote: Repository not found.
fatal: repository 'https://github.com/sap-labs-france/ev-aws.git/' not found
i have pasted this url on chrome "https://github.com/sap-labs-france/ev-aws.git/" but it seems that doesnt exist anymore and i obtain a git 404.
Please can you help me?

LucasBrazi06, 2021-09-20:
That's a normal error, this is a private repo.
You can ignore it.

niahane, 2021-09-20:
It ask me for user and password, if i skip hitting enter key it not go forward.
I had to jump the make server step?

LucasBrazi06, 2021-09-20:
Have you checked the file ./ev-server/docker/buildspec.yml?

LucasBrazi06, 2021-09-24:
This is more an issue in dealing with sub-modules (Git) in Docker.

closed 2021-09-24 by LucasBrazi06