sap-labs-france/ev-server #2862: Documentation: Is there an API reference doc for the REST Central Server?

Pulse-akhil, 2021-09-25:
It will be great to have the REST API documentation for the OCPP and OCPI functionalities provided by the ev-server REST server. Is this readily available

LucasBrazi06, 2021-09-25:
We have a Swagger UI:
http://localhost:8080/v1/docs/#/

closed 2021-09-25 by LucasBrazi06

Pulse-akhil, 2021-09-26:
Do I have to enable something in config.json to enable this port?  I don't see this port open in the list of active ports opened by ev-server app.

LucasBrazi06, 2021-09-26:
> Do I have to enable something in config.json to enable this port? I don't see this port open in the list of active ports opened by ev-server app.

No, just put the URL/Port of your REST server + '/v1/docs'

Pulse-akhil, 2021-09-26:
Thanks, looks like the assets switch to https:// by default. I am using an nginx in front of the ev-server to route incoming requests to the right port. Is there a way to make the protocol the default one that the browser requests for?

Pulse-akhil, 2021-09-26:
Please ignore, looks like it's Chrome that's behaving this way. The urls you have are relative.