sap-labs-france/ev-server #2879: Server with docker missing files

danielrama30, 2021-09-28:
Hi, we´re trying to start the server with docker but we get some errors.
-Missing componentExport.js file with latest versions
-with version 2.4.88 we can setup and start the container but the server does not connect to mongodb which was installed and configured by the server itself.
-the container sometimes stops randomly

danielrama30, 2021-09-28:
Another problem which it´s not explained in the installation instructions. How do we configure the emails with the docker installation?

LucasBrazi06, 2021-09-28:
componentExport.js does not exist anymore, where did you get this file?

danielrama30, 2021-09-29:
> componentExport.js does not exist anymore, where did you get this file?
> Maybe you package.json is not in sync with the latest version.

that´ the point there is not componentExport.js but the installation process gives us error because it´s missing. with version 2.4.88 with has the file doesn´t give any problem. the package.json is the one which comes with the relative versions.

we managed to fix connection problem tomongodb, was a bad configuration, and docker apparently stopped stopping the container at random

LucasBrazi06, 2021-09-29:
> that´s the point there is not componentExport.js but the installation process gives us error because it´s missing from version 2.4.90 to 2.4.92. Instead with version 2.4.88 which has the file doesn´t give any problem. the package.json are the ones which comes with the relative versions.

Okay, the call to componentExport.js was done by the package.json launch script and I cleaned it up.
I cannot find any occurrences of this file anymore anywhere.
Why do you still have it!?

> we managed to fix connection problem tomongodb, was a bad configuration, and docker apparently stopped stopping the container at random

GUT!

niahane, 2021-09-29:
Hi Lucas, it seems there is another problem in the latest docker git version, in a fresh docker environment i have do this:
git clone https://github.com/sap-labs-france/ev-server.git && cd ev-server/docker
make SUBMODULES_INIT=false
At the end i obtain:
Hash: 445bede7471971e9345b
Version: webpack 4.46.0
Time: 405ms
Built at: 09/29/2021 9:27:13 AM
 338 assets
Entrypoint main = start.js start.js.map
[0] ./src/start.ts 299 bytes {0} [built] [failed] [1 error]

ERROR in ./src/start.ts
Module build failed (from ./node_modules/ts-loader/index.js):
TypeError: loaderContext.getOptions is not a function
    at getLoaderOptions (/usr/builder/node_modules/ts-loader/dist/index.js:91:41)
    at Object.loader (/usr/builder/node_modules/ts-loader/dist/index.js:14:21)
npm ERR! code ELIFECYCLE
npm ERR! errno 2
npm ERR! ev-server@2.5.1 build:prod: `webpack --env prod`
npm ERR! Exit status 2
npm ERR! 
npm ERR! Failed at the ev-server@2.5.1 build:prod script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-09-29T09_27_13_677Z-debug.log
ERROR: Service 'server' failed to build: The command '/bin/sh -c npm run build:${build}' returned a non-zero code: 2
make: *** [Makefile:28: server] Errore 1

The log file in /var/lib/docker/overlay2/2769e9cfd8685f61d2b5529caeeda9f1d2fa2ff6023ccc6acf25a8f1f1f12036/diff/root/.npm/_logs/2021-09-29T09_27_13_677Z-debug.log:

0 info it worked if it ends with ok
1 verbose cli [ '/usr/local/bin/node', '/usr/local/bin/npm', 'run', 'build:prod' ]
2 info using npm@6.14.15
3 info using node@v14.18.0
4 verbose run-script [ 'prebuild:prod', 'build:prod', 'postbuild:prod' ]
5 info lifecycle ev-server@2.5.1~prebuild:prod: ev-server@2.5.1
6 info lifecycle ev-server@2.5.1~build:prod: ev-server@2.5.1
7 verbose lifecycle ev-server@2.5.1~build:prod: unsafe-perm in lifecycle true
8 verbose lifecycle ev-server@2.5.1~build:prod: PATH: /usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/usr/builder/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
9 verbose lifecycle ev-server@2.5.1~build:prod: CWD: /usr/builder
10 silly lifecycle ev-server@2.5.1~build:prod: Args: [ '-c', 'webpack --env prod' ]
11 silly lifecycle ev-server@2.5.1~build:prod: Returned: code: 2  signal: null
12 info lifecycle ev-server@2.5.1~build:prod: Failed to exec build:prod script
13 verbose stack Error: ev-server@2.5.1 build:prod: `webpack --env prod`
13 verbose stack Exit status 2
13 verbose stack     at EventEmitter.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:400:28)
13 verbose stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:400:28)
13 verbose stack     at maybeClose (internal/child_process.js:1058:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:293:5)
14 verbose pkgid ev-server@2.5.1
15 verbose cwd /usr/builder
16 verbose Linux 5.11.0-34-generic
17 verbose argv "/usr/local/bin/node" "/usr/local/bin/npm" "run" "build:prod"
18 verbose node v14.18.0
19 verbose npm  v6.14.15
20 error code ELIFECYCLE
21 error errno 2
22 error ev-server@2.5.1 build:prod: `webpack --env prod`
22 error Exit status 2
23 error Failed at the ev-server@2.5.1 build:prod script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 2, true 

There is something that i can do for fix this problem ?

LucasBrazi06, 2021-09-29:
Good catch thanks!
This is a regression.
I will upgrade to webpack 5 and let you know.

LucasBrazi06, 2021-09-29:
Came across also the error on componentExport :-)
Fixing it

niahane, 2021-09-29:
Thanks to you for the support, it is difficult to find which parts of the guide in readme should be followed in docker mode, I understood after that the files in the docker folder should be configured first.
Now i just assembled a new debian 11 server to do some testing, i redid all the steps but it still hangs in the usual spot. Let me know when you have fixed the code that I will gladly try again, in the meantime I put version 2.8.44 back online.
For now I haven't been able to get the emails working yet, but I solved the recaptcha problem and with version 2.4.88, by stealing the string from the email body from robo3t I can now create the subdomains.

LucasBrazi06, 2021-09-29:
My pleasure :-)
I just pushed a commit.
Could you please check on master-qa and let me know?

LucasBrazi06, 2021-09-29:
> For now I haven't been able to get the emails working yet, but I solved the recaptcha problem and with version 2.4.88, by stealing the string from the email body from robo3t I can now create the subdomains.

Here is our conf for the email:

[image]

niahane, 2021-09-30:
Hi Lucas, good morning. I have just tested server and dashboard, compilation works fine on them but now the dashboard cant'connect to server. Now i am try a previous version for see if the problem appear again or there is some trouble with the recent changes.

LucasBrazi06, 2021-09-30:
> Hi Lucas, good morning. I have just tested server and dashboard, compilation works fine on them but now the dashboard cant'connect to server. Now i am try a previous version for see if the problem appear again or there is some trouble with the recent changes.

The connection from the front-end to the backend is defined in the config.json file of the front-end.

niahane, 2021-09-30:
Yes i know, this is my scenario:
I have a static dns that point in my office public ip for outside access, in my router i have added a static dns record to evserver ip evserver.mydomain.cloud -> 192.168.1.105 (in this server there is ev-server and ev-dashboard ) for access to server when i am connected to the office network.
In the config.json i have replaced all "server" recurrence  host with ip of the interface ethernet of the server, i also tried "localhost" or "mydnsname.mycompanyname.org" nothing seems to work if i install evserver and evdashboard on this pc i usning for writing now and i open http://mypcip:3080 login seems to work well dashboard can connect to server running on the same pc without problems . In docker mode which host should I put in the configuration? I had to bind docker internal network server ip? 172.18.0.5? or ip of the server running docker 192.168.1.105?

LucasBrazi06, 2021-09-30:
Welcome to my world :D !!!
This is why we have setup a cloud :-)
You can check the logs in the DB with Robot 3T or any other tools.
Check your console when signing on (pb with reCaptcha?)

danielrama30, 2021-09-30:
Hello, I've been following the conversation even if didn't reply, but I'm getting an error after another trying to make it reachable from the internet, and never made it with docker even to reach it locally from another computer in the same network. Maybe Can you leave a copy or something of the fully functional docker containers, just hiding the sensible informations like public ips and passwords, because I really don't know what else to do.

LucasBrazi06, 2021-09-30:
> Hello, I've been following the conversation even if didn't reply, but I'm getting an error after another trying to make it reachable from the internet, and never made it with docker even to reach it [...]

We are using our docker scripts in our AWS Fargate clusters and I unfortunately cannot provide support on a cloud that you have built otherwise I would be overwhelmed with issues, sorry for this!

danielrama30, 2021-10-01:
ok, then can I ask you at least to write comments of what every string in the configs does. There are some things that we do not understand if it should go on the internet or should stay in localhost, since if we set something wrong the dashboard will not communicate with the central server.

LucasBrazi06, 2021-10-01:
> ok, then can I ask you at least to write comments of what every string in the configs does? There are some things that we do not understand if it should go on the internet or should stay in localhost, since if we set something wrong the dashboard will not communicate with the central server.

Yes this will be documented with a JSon schema but not in short term unfortunately.

niahane, 2021-10-01:
Good morning, no problem we understand, and sorry to disturb with all these requests we usually try not to bother and do it alone but it is a complex infrastructure, so let's take advantage of this space to ask you as many questions as possible to try to do less in the future. 
Assuming that we have a static dns for outside access and inside (my office) access

evserver.con9.cloud point to a pc (192.168.1.100) running ubuntu-server it has docker, and inside docker there is ev-server (172.18.0.5/16) and ev-dashboard (172.18.0.6/16) containers.

I think this is the most relevant part of the server configuration:

`{
  "Crypto": {
    "key": "ShVmYq3t6w9z$C&F)H@McQfTjWnZr4u7",
    "algorithm": "aes-256-ctr"
  },
  "EVDatabase": {
    "url": "https://ev-database.org/export_v26/74964691",
    "key": ""
  },  
  "CentralSystems": [
    {
      "type": "ocpp",
      "implementation": "soap",
      "protocol": "http",
      "host": "server", <-what mean "server" i had to point it on outside server? I had to bind it on static dns or ip of internal container or ip of machine running docker or is an internal variable?
      "port": 8000,
      "debug": false
    },
    { 
      "type": "ocpp",
      "implementation": "json",
      "protocol": "ws",
      "host": "server", <- like above
      "port": 8010,
      "debug": false
    }
  ],  
  "CentralSystemRestService": {
    "protocol": "http",
    "host": "server", <- like above
    "port": 81,
    "userTokenKey": "s3A9js135sdf384hgj384ku38lilckjflskdjf2797boeiBhxQDM1GInRith",
    "userTokenLifetimeHours": 12,
    "userDemoTokenLifetimeDays": 365,
    "userTechnicalTokenLifetimeDays": 365,
    "socketIO": true,
    "socketIOListNotificationIntervalSecs": 5,
    "socketIOSingleNotificationIntervalSecs": 1,
    "passwordWrongNumberOfTrial": 3,
    "passwordBlockedWaitTimeMin": 5,
    "captchaSecretKey": "6LeIxAcTAAAAAGG-vFI1TnRWxMZNFuojJ4WifJWe",
    "debug": false
  },  
  "CentralSystemFrontEnd": {
    "protocol": "http",
    "host": "evserver.con9.cloud", <- i pointed it on static dns i think is a declaration for server to communicate with dashboard right? Pointing it on my static dns and from dashboard config section CentralSystemServer and FrontEnd host "evserver.con9.cloud" i am able to login in the dashboard without problem, is the right way for configuring it?
    "port": 3080
  },
`
Thanks in advance

LucasBrazi06, 2021-10-01:
> "host": "server", <-what mean "server" i had to point it on outside server? I had to bind it on static dns or ip of internal container or ip of machine running docker or is an internal variable?

I always let the 'host' empty for the servers (REST, Json, Soap...)

> "host": "evserver.con9.cloud", <- i pointed it on static dns i think is a declaration for server to communicate with dashboard right? Pointing it on my static dns and from dashboard config section [...]

This one is to identify where is the front-end to build URLs that we send in notification emails.

niahane, 2021-10-01:
It's incredible, on my pc (kde neon) docker + evserver + evdashbord everything seems to work. I do the same configuration on a debian server identical to this one, I adapt the dns to the new ip, with nslookup I verify that everything is ok, nothing the dashboard cant'connect to server. After days of frustation and rage i declare KO.

LucasBrazi06, 2021-10-01:
> It's incredible, on my pc (kde neon) docker + evserver + evdashbord everything seems to work i am able to connect also the simulator, i can see dummy av chargers, abb, siemens etc.-. I do the same configuration on a debian server identical to this one, I adapt the dns to the new ip, with nslookup I verify that everything is ok, nothing the dashboard cant'connect to server, i set up a new machine ubuntu server nothing dashboard cant connect... After days of frustation and rage i declare KO.

Never let it down :-)

closed 2021-10-01 by LucasBrazi06