sap-labs-france/ev-server #2880: Emails from server.

niahane, 2021-09-28:
How do we configure the emails with the docker installation? In the docs there is not any reference. We are stuck with creating a subdomain that require email confirm but email never received, is hard to find the logs without any instrucion the mail has stopped work in the server application? In the dashboard or in maildev?

niahane, 2021-09-28:
I have connected robo3t to mongo db i find the acrivation url in the email body logs i have pasted in a browser:
http://test.localhost:3080/verify-email?VerificationToken=d5477a819eeb3 etc.. etc..&Email=(email used in the registration process)&ResetToken=727eec0d-6705-44e6-b3c8-1d50d6a24e1d.
The result page seems to work it ask me for a password change, i set a new password and press save but nothing happens no error log, i am tring to login with the inserted password but "username or password error"

niahane, 2021-09-28:
Ok i have found the configuration file config.json inside the container i have modified it with my gmail login. saved and restarted container, i am tryng to send email but it still not work.  "No suitable backup SMTP server configuration found or activated to send email after an error on the main SMTP server",
I have do a full reset of my docker system cloned server git from this site, modified config.json inside docker folder before deploy, ran make SUBMODULES_INIT=false (if i omit submodule_init it ask me for git username and password and stuck because git not support user and password login anymore)
Nothing... same error
The relative section of config.json:
  "Email": {                   
    "from": "con9fax@gmail.com",  
    "admins": [                               
      "supporto@con9.it"                                                           
    ],                           
    "smtp": {                        
      "from": "con9fax@gmail.com",                       
      "host": "smtp.gmail.com",               
      "port": 465,    (also tried 587 no make difference)                         
      "secure": true,                
      "requireTLS": true,           
      "type": "login",                                             
      "user": "con9fax",       
      "password": "mypassword",
      "debug": false            
    }

LucasBrazi06, 2021-09-28:
> The result page seems to work it ask me for a password change, i set a new password and press save but nothing happens no error log, i am tring to login with the inserted password but "username or password error"
> If i try to reset password it doesnt work if i press reset nothing happens, no error, no confirmation no log update.

Check the console log in the browser, you should have an issue with the reCaptcha configuration (version V3)

> "No suitable backup SMTP server configuration found or activated to send email after an error on the main SMTP server",

If the primary SMTP server conf fails, we try with the secondary which is not provided I think in your case.
Try to search for the previous error on the primary email conf you provided in your comment.
Let me know.