sap-labs-france/ev-server #3013: Invoices are not generating for payment

rccess, 2021-10-30:
Hello team,

We added stripe , when new user getting signup , profile creating on stripe 

[image]

[image]

what may be the reasons

below is the defined in config file for schedule task

[image]

LucasBrazi06, 2021-10-30:
@ClaudeROSSI Could you check?
Thanks!

ClaudeROSSI, 2021-10-31:
Hello,

@rccess : Can you provide more details about the stripe configuration?
On which tenant are you facing the problem?

Regards,
Claude

rccess, 2021-10-31:
invoice are not generating for any of the tenant and user 

[image]

ClaudeROSSI, 2021-10-31:
I'm afraid, the information provided is not sufficient. Which customer accounts (which tenant) are you talking about.
Please send me the URL where to logon! I'll then register myself and ask you to grant me the admin role. Only then I'll be able to investigate.

Claude

rccess, 2021-10-31:
okay

rccess, 2021-11-01:
@ClaudeROSSI  i have activated your account and changed the permission to admin

ClaudeROSSI, 2021-11-01:
Thanks. I now have access to the logs.

At a first glance it seems that the currency set in the "> integration settings > pricing module" does not match the one set in your stripe account. Can you please double check? We need the same currency on both sides!

Please also confirm that a few customers have been created on the STRIPE side. 
Be aware that STRIPE customers are created on-demand (i.e.: each time a user enters a payment method). So you should see there only the ones that have been able to enter their credit card information.

Let me know if you need more information.
Thanks,
Claude

ClaudeROSSI, 2021-11-01:
Please also make sure you are properly connected to the STRIPE productive account. One common mistake is to have eMobility pointing to the strip test account (i.e.: by entering a wrong secret key in the BILLING settings).
Best regards,
Claude

rccess, 2021-11-01:
thanks for the update

i do have INR in Stripe account (it is in production mode) as well updated same in dashboard

[image]

but invoice is generating yet

[image]

[image]

you can check last charging session

thanks

ClaudeROSSI, 2021-11-03:
Hello,

All sessions have been made on a pubic site. No invoices are generated when the site is public.
Please make sure to uncheck that option to actite the billing module!

[image]

Le me know if you have any concerns!

rccess, 2021-11-03:
thanks for your update

i have updated the sites without public 

[image]

and started 2 sessions but still no invoice generated yet

[image]

ClaudeROSSI, 2021-11-03:
I have a question regarding the scheduled tasks.
Why did you change the settings for the task: **SynchronizeBillingInvoicesTask**

This task is deprecated. Please switch it OFF like in the example below:

      {
        "name": "SynchronizeBillingUsersTask",
        "active": true,
        "periodicity": "0 0 * * *",
        "config": {}
      },
      {
        "name": "BillingPeriodicOperationTask",
        "active": true,
        "periodicity": "0 3 * * *",
        "config": {}
      },
      {
        "name": "SynchronizeBillingInvoicesTask",
        "active": false,
        "periodicity": "0 0 * * *",
        "config": {}
      },

rccess, 2021-11-03:
i have changed the config.json as suggested 

[image]

still not generating invoice for long session as well !

[image]

rccess, 2021-11-04:
issued fixed 

"AsyncTask": {
    "active": true,
    "nbrTasksInParallel": 4
  },

it was false in config file

thanks for your help

closed 2021-11-04 by rccess