sap-labs-france/ev-server #3019: mobile app scanning to add tenant is not working

rccess, 2021-11-03:
Hello team,

mobile app scanning to add tenant is not working

Showing Error Unknown Cloud 'AWS'

[image]

below is the configuration 

[image]

LucasBrazi06, 2021-11-03:
@AlixH Could you check?

AlixH, 2021-11-05:
The configuration shoud be as follow : 
```
public static readonly DEFAULT_ENDPOINT_CLOUD_ID = 'aws';
  public static readonly ENDPOINT_CLOUDS: EndpointCloud[] = [
    { id: Configuration.DEFAULT_ENDPOINT_CLOUD_ID, name: 'Amazon Web Service', endpoint: Configuration.AWS_REST_ENDPOINT_PROD }
  ];
```

The error is shown because the DEFAULT_ENDPOINT_CLOUD_ID is 'app' but should be 'aws'  (encoded in the QR-Code)

LucasBrazi06, 2021-11-18:
Closed for inactivity.
Let me know if you still have the issue.
Thanks!

closed 2021-11-18 by LucasBrazi06