sap-labs-france/ev-server #3034: Mail not receiving properly

rccess, 2021-11-06:
Hello team,

On simulation of Charging stations mails received to admin is not in proper conditions. Please look into once below

[image]

rccess, 2021-11-10:
any chance to get the reply on this issue ?

LucasBrazi06, 2021-11-10:
What do you me by proper condition?

rccess, 2021-11-10:
i Need to send the issue in detail ?

LucasBrazi06, 2021-11-10:
I don't know what's wrong with the email ;)

rccess, 2021-11-10:
The Notification comes to admin when Charging Station get Connected or Registered 

The Email Template not coming properly, not html loading, original raw email coming to admin mail id

issue might be in https://github.com/sap-labs-france/ev-server/blob/c603f935ef3c33ae5006c4dfe1de05115dbdbf93/src/notification/email/EMailNotificationTask.ts#L79

This is the screenshot below

[image]

LucasBrazi06, 2021-11-10:
Ah ok, I got it.
@rmangin-fr Could you have a look and test the email notif on a GMail account?
Thanks!

rmangin-fr, 2021-11-10:
Tested with no reproducible case.
GMail shows the ChargingStationRegistered notification as expected
See screenshot bellow:
[image]

rmangin-fr, 2021-11-10:
German language tested as well with no error:
[image]

rmangin-fr, 2021-11-10:
Hi,
Here is the English evidence of not reproducible:
[image]

rccess, 2021-11-10:
Smtp compatibility might be ?

Can you please share your config file? Might be i am missing something
there ?

LucasBrazi06, 2021-11-18:
Not a lot to share:

[image]

LucasBrazi06, 2021-11-25:
Let me know if your problem is solved.

closed 2021-11-25 by LucasBrazi06