sap-labs-france/ev-server #3048: OCPP => Central System connection/registration issue

archerixx, 2021-11-10:
Hi,
We are trying to connect Charging Station's OCPP to ev-server Central System using http and/or websocket, but OCPP does not recognise ev-server as Central System even though it does receive POST/GET request from Charing Station.

When using websocket and OCPP v1.6 with following configuration:
```
  "CentralSystems": [
    {
      "implementation": "soap",
      "protocol": "ws",
      "host": "192.168.163.122",
      "port": 8001
    }
  ]
```

We are getting this GET request response:
```
	
[
      {
            "url": "/QssCharger",
            "method": "GET",
            "query": {},
            "body": {},
            "locale": "en_US",
            "xhr": false,
            "ip": "192.168.163.125",
            "ips": [],
            "httpVersion": "1.1",
            "headers": {
                  "connection": "Upgrade",
                  "host": "192.168.163.122:8001",
                  "sec-websocket-key": "Nw3nPFxVbn5bKgX3mzwB3g==",
                  "sec-websocket-protocol": "ocpp1.6",
                  "sec-websocket-version": "13",
                  "upgrade": "websocket",
                  "user-agent": "WebSocket++/0.7.0"
            }
      }
]
```
If using JSON instead of SOAP, we get "Unauthorized" response and message on Frontend: 

> Invalid connection URL /QssCharger

When using http and OCPP v1.5 with following configuration:
```
  "CentralSystems": [
    {
      "implementation": "soap",
      "protocol": "http",
      "host": "192.168.163.122",
      "port": 8000
    }
  ]
```
We are getting this POST request response:
```
[
      {
            "url": "/",
            "method": "POST",
            "query": {},
            "body": {
                  "SOAP-ENV:Envelope": {
                        "$": {
                              "xmlns:SOAP-ENV": "http://www.w3.org/2003/05/soap-envelope",
                              "xmlns:SOAP-ENC": "http://www.w3.org/2003/05/soap-encoding",
                              "xmlns:xsi": "http://www.w3.org/2001/XMLSchema-instance",
                              "xmlns:xsd": "http://www.w3.org/2001/XMLSchema",
                              "xmlns:cp": "urn://Ocpp/Cp/2012/06/",
                              "xmlns:cs": "urn://Ocpp/Cs/2012/06/",
                              "xmlns:wsa5": "http://www.w3.org/2005/08/addressing"
                        },
                        "SOAP-ENV:Header": [
                              {
                                    "cs:chargeBoxIdentity": [
                                          {
                                                "_": "default",
                                                "$": {
                                                      "SOAP-ENV:mustUnderstand": "true"
                                                }
                                          }
                                    ],
                                    "wsa5:MessageID": [
                                          "urn:uuid:1c3851ec-02e1-41cf-bf25-4bcc96ab3a95"
                                    ],
                                    "wsa5:From": [
                                          {
                                                "wsa5:Address": [
                                                      "http://qsscharger.com:50000"
                                                ]
                                          }
                                    ],
                                    "wsa5:To": [
                                          {
                                                "_": "http://192.168.163.122:8000",
                                                "$": {
                                                      "SOAP-ENV:mustUnderstand": "true"
                                                }
                                          }
                                    ],
                                    "wsa5:Action": [
                                          {
                                                "_": "/BootNotification",
                                                "$": {
                                                      "SOAP-ENV:mustUnderstand": "true"
                                                }
                                          }
                                    ]
                              }
                        ],
                        "SOAP-ENV:Body": [
                              {
                                    "cs:bootNotificationRequest": [
                                          {
                                                "cs:chargePointVendor": [
                                                      "Circontrol"
                                                ],
                                                "cs:chargePointModel": [
                                                      "eVolve Smart TM4"
                                                ],
                                                "cs:chargePointSerialNumber": [
                                                      "12033029170001"
                                                ],
                                                "cs:firmwareVersion": [
                                                      "3.4.0 rc1"
                                                ]
                                          }
                                    ]
                              }
                        ]
                  }
            },
            "locale": "en_US",
            "xhr": false,
            "ip": "192.168.163.125",
            "ips": [],
            "httpVersion": "1.1",
            "headers": {
                  "host": "192.168.163.122:8000",
                  "user-agent": "gSOAP/2.8",
                  "content-type": "application/soap+xml; charset=utf-8; action=\"/BootNotification\"",
                  "content-length": "1161",
                  "connection": "close",
                  "accept-encoding": "gzip, deflate",
                  "soapaction": "\"/BootNotification\""
            }
      }
]
```
When using JSON, we get "Bad request".

On OCPP (Charging Station), CB is never registered successfully to Central system so complete connection is never made.
I have tested with different Central System and was able to make connection, so I am sure that configuration on OCPP are correct.

Would you be able to let us know if we should be using different Central system configuration and if there are maybe some other factors that should be looked into when making connection with OCPP ?

Thank you in advance

LucasBrazi06, 2021-11-10:
Try this one:
```
		{
			"type": "ocpp",
			"implementation": "json",
			"protocol": "ws",
			"host": "<YOUR_HOST>",
			"port": 8001,
			"debug": false
		}

```

archerixx, 2021-11-10:
I get same GET response, but when CS tries to send responde it gets 401 "Unauthorized" then connection closes. Same message on frontend.

> Invalid connection URL /QssCharger

In CLI this is visible:
```
OCPP-J Server listening on 'ws://192.168.163.122:8001'
```

LucasBrazi06, 2021-11-10:
You have to generate a token and copy the link in the UI.
We don't allow any charger to connect to the backend.

archerixx, 2021-11-10:
How can a Token be generated and where ?
And to which UI does have to be copied in ?

Are we talking about "userTokenKey" or some other ?

archerixx, 2021-11-10:
I have it installed

LucasBrazi06, 2021-11-10:
[image]

archerixx, 2021-11-10:
[image]

I am missing those tabs in dashboard UI. I have created Organization with all components, which lead me to this page. Should I create it in some other order or run/start Dashboard differently ?

I have checked other Issues on this repository and found similar issue on [2099](https://github.com/sap-labs-france/ev-server/issues/2099). But unfortunately it was dead end.

I can see that token is mentioned on issue [658](https://github.com/sap-labs-france/ev-server/issues/658), but it does not say how it is generated.

LucasBrazi06, 2021-11-10:
It's because you are not an Admin ;-)
Change your user's role in the DB from 'B' to 'A'.

archerixx, 2021-11-11:
Thank you for your help so far. I have just a few more questions and hopefully everything will work after it.

I had user on role 'S' :')
After changing to 'A', I can see page and can create token succesfully, but unfortunatelly 'Copy' icon is blank (not working) so I could not copy anything from there.

In database I found token details:
```
{
  siteAreaID: null,
  description: 'Test',
  expirationDate: '2021-12-11T09:55:56.715Z',
  createdBy: { id: '6185459525d1e2c52f84e9b2' },
  createdOn: 2021-11-11T09:56:00.723Z,
  id: '618ce8b03c2be94bdfbde183'
}
```

Does all this represent token ? 

> You have to generate a token and copy the link in the UI. We don't allow any charger to connect to the backend.

Which link needs to be copied and where exactly should it be copied to ?

LucasBrazi06, 2021-11-11:
> I had user on role 'S' :')

Role 'S' is Super Admin for creating organizations (or tenant or customer namespace) when accessing the dashboard without sub-domain (eg. http://localhost/).

Once the org is created you have access to it (eg. http://<ORG_SUB_DOMAIN>.localhost/) and here you have the role Admin ('A'), Basic ('B') and Demo ('D').

> Does all this represent token ?

Yes, this is the token, but if you provided the property in the config.json file, you should have had it in the UI.

It's of form:

`wss://<Ocpp-j-host>/OCPP16/<OrgID>/<TokenID>`

In your case:
`wss://192.168.163.122:8001/OCPP16/<OrgID>/618ce8b03c2be94bdfbde183 `

archerixx, 2021-11-11:
Thank you very much for your help, I managed to get it working.

LucasBrazi06, 2021-11-11:
You welcome :-)

closed 2021-11-11 by LucasBrazi06