sap-labs-france/ev-server #3049: Json server not running?

terryjoshua, 2021-11-12:
HI, 

I am working on charging station connection to central server by OCPP-J.
I was able to create token URL thanks to Fabiano's help.
But I still could not make "charging station" connect to central server.

I used to see this log when I launched ev-server module.

> **OCPP-J Server listening on 'ws://xx.xx.xx.232:8010'**

But not any more, Here is log output when I launched ev-server.

**> 11/12/2021, 8:51:30 PM - Server is starting...
> 11/12/2021, 8:51:30 PM - Starting Rest Server...
> 11/12/2021, 8:51:30 PM - Starting Soap Server...
> 11/12/2021, 8:51:30 PM - Starting Json Server...
> 11/12/2021, 8:51:30 PM - Starting Ocpi Server...
> 11/12/2021, 8:51:30 PM - Starting Oicp Server...
> 11/12/2021, 8:51:30 PM - Starting OData Server...
> 11/12/2021, 8:51:30 PM - Server Rest, Soap, Json, Ocpi, Oicp, OData has been started successfully in 0.044 secs
> 11/12/2021, 8:51:30 PM - Scheduler is starting...
> 11/12/2021, 8:51:30 PM - Ocpi Server listening on 'http://xx.xx.xx.232:9090'
> 11/12/2021, 8:51:30 PM - Soap Server listening on 'http://xx.xx.xx.232:8000'
> 11/12/2021, 8:51:30 PM - Oicp Server listening on 'http://xx.xx.xx.232:9080'
> 11/12/2021, 8:51:30 PM - Rest Server listening on 'http://xx.xx.xx.232:82'
> 11/12/2021, 8:51:30 PM - OData Server listening on 'http://xx.xx.xx.232:9292'
> 11/12/2021, 8:51:30 PM - Scheduler has been started successfully in 0.025 secs
> 11/12/2021, 8:51:30 PM - Async Task manager is starting...**

OCPP-J server is not running? I verified server did listen 8010 port by "ss -nat" command.
But I could not see "Json Server listening" log message any more.
Here is json part from config.json file.

**> {
>       "type": "ocpp",
>       "implementation": "json",
>       "protocol": "ws",
>       "host": "xx.xx.xx.232",
>       "port": 8010,
>       "keepaliveinterval": 30,
>       "debug": false
>     }
>** 

Should I specify "wss" in protocol? Since I heard that only secure connection was supported.
I saw the following errors from defaults.logs. (This is only error log.)

**> [
>       {
>             "error": "Error: ENOENT: no such file or directory, open '/home/terry/ev-server/dist/assets/charging-station-templates/charging-stations.json'
>    at Object.openSync (node:fs:585:3)
>    at Object.readFileSync (node:fs:453:35)
>    at Function.updateChargingStationTemplatesFromFile (/home/terry/ev-server/dist/webpack:/ev-server/src/start.ts:170:48)
>    at Function.start (/home/terry/ev-server/dist/webpack:/ev-server/src/start.ts:142:18)"
>       }
> ]
>** 

Any help will be greatly appreciated.

Thanks,

LucasBrazi06, 2021-11-14:
> I was able to create token URL thanks to Fabiano's help.

My first name is Serge :-)

> But not any more, Here is log output when I launched ev-server.

I have rewritten the WS Server using uWebSocket.
Could you give it another try?
It should start right away before the others based on ExpressJs:

[image]

terryjoshua, 2021-11-14:
Thanks a lot. I am so sorry to mispronounce your name.
I can see the line : Json Server listening on http:// line with updated code. Thanks~~~~
But I expected ws://xx.xx.xx.xx:8010/ line.

It should be OK even though "protocol ws" should be used?

I will let other guy test new Json connection as soon as possible.

Thanks,

LucasBrazi06, 2021-11-14:
> Thanks a lot. I am so sorry to mispronounce your name.

No problem at all :-) !!!

> But I expected ws://xx.xx.xx.xx:8010/ line.
> It should be OK even though "protocol ws" should be used?

It's 'wss' over 'https' via an upgrade of the protocol so it's okay with ws:// (http) or wss:// (https).
We listen to the HTTP port because usually in the cloud, the HTTPs is managed by the load balancer and within the cluster we use http.

> I will let other guy test new Json connection with sample charging station and let you know the result.

Okay!

terryjoshua, 2021-11-15:
It is working only by ws:// protocol. we could not connect to server by wss:// protocol.
It is OK. At least, we know we can connect to central server from our charging station.
I am running some "feasibility test", now. So far so good.
I will move on mobile client (android/ios) test.

LucasBrazi06, 2021-11-15:
You welcome!

FYI, wss:// is over HTTPs and managed by the load balancer in production:

Charging Station wss -> LB https/wss -> Backend http/ws

closed 2021-11-15 by LucasBrazi06