sap-labs-france/ev-server #3086: Implementing Stop Charging Transaction on the basis of 'currentCumulatedPrice' automatically

rccess, 2021-11-19:
Hello team,

Hope you are doing great !
We are trying to something different 
**Reason to do :** we are trying to implement pre-paid charging services for Open-mobility users.

**Case :** We have assigned virtual Wallet to each user and giving the option to add money into wallet i,e $10,$20,$50 etc...
Desire : We want to that user is able to start charging if user have positive balance in their wallet and stop charging automatically if user wallet balance reach to 0. (Which we implemented already with update on each transaction.storage)

**Challenge :** We are not able to call Stop Charging since stop charging function takes user temporary token to stop transaction and since the stop function will need by the system if the wallet balance is equal to '0'

**Need Advise** : We want to know how we can call stop function without token. if we can have small idea from your site, we will implement the same.

Hope you will help and will advise on this concern. 

Thanks in Advance

LucasBrazi06, 2021-11-19:
Thanks for your issue!
Yes, we are working like hell :-), I hope you are fine too?
Prepaid scenario is something we wanted to address also on our side.
What about reviewing what you have done and integrating it into the standard?

For the Stop Transaction, you can create a special user with role Admin and assign him a specific RFID card which you will always use to stop the transaction.

ravikarodiya, 2021-11-19:
Thanks for your reply.
When i am calling the curl from TransactionStorage file's saveTransaction function to stop the transaction it give's error because of wrong accessToken but in same function i don't have login details i only have the userID on that function. so how can i pass accessToken in that

rccess, 2021-11-19:
@ravikarodiya Can we pass RFID Card with put Static RFID Card details on function call time (as @LucasBrazi06 suggested the same), which will give the permission to stop the transaction, 
but the question is how Admin user accessToken will pass (if is compulsory ) since it generate on login time from interface 

What you think @LucasBrazi06

LucasBrazi06, 2021-11-19:
> Thanks for your reply. When i am calling the curl from TransactionStorage file's saveTransaction function to stop the transaction it give's error because of wrong accessToken but in same function i [...]

Ah, my bad, I confused the Token (RFIDCard) with the WebSocket (WS) Token :-(, sorry!
You are talking about sending a command from the UI -> REST server -> Json Server -> Charging Station, right?

The WS Token is the one we use to register a new Charging Station (CS), it is stored along the CS in the DB.
Whenever you want to call the CS (REST server -> Json Server), we open a WS connection between the REST and the Json server where the CS are connected and you need to pass this token in the URL connection.

WS URL : /REST/<TENANT_ID>/<TOKEN_ID>/<CHARGING_STATION_ID>

Let me know if this solves your problem.

rccess, 2021-11-19:
Thanks for your reply ! 

we are calling like 

PUT /v1/api/charging-stations/CS-xxxxxxxEVCS-0002/remote/start - 200 - 1224.642ms - 0.0205078125Kb

able to start like this but not able to stop like this since user accessToken not able call and we want to hit like this url which stop charging station and put it in normal same as like we do manually with stop button

PUT /v1/api/charging-stations/CS-xxxxxxxEVCS-0002/remote/stop

but this url not able to hit hit without user accessToken (not Charging Token)

Please let me know your idea, since you are expert of it.

LucasBrazi06, 2021-11-19:
Could you provide me the error that you have in the logs?

rccess, 2021-11-19:
PUT /v1/api/charging-stations/CS-xxxxxxxEVCS-0002/remote/stop 

getting access denied due to not passing accessToken of user since this url not hitting by user, this request url hitting by system with the value of currentCumulatedPrice is increase with wallet balance

is there any way by which we can pass this request with  putting any static token 

Please do advise , 

@ravikarodiya will share the logs also

LucasBrazi06, 2021-11-19:
Ah, okay! Got it!!! LOL!!!

No, this is a protected API, so you cannot call this endpoint without being authenticated and pass a JWT token!

No static token either, but I see one solution which can introduce a small security breach:

You'll have to create a dedicated user and mark it Technical:
[image]

For Technical Users, which are used for B2B communications (user cannot login using the front-end), the token lasts for a year but you can change it in the config.json file: CentralSystemRestService -> userTechnicalTokenLifetimeDays.
-> Set this value to 100 years and you have your static token :-).

Log into the app, get back the token and use it in all your HTTP requests.

Static token, if they got captured. will give access to the backend forever until you change the key in the config.json file (userTokenKey.)

Voila!

closed 2021-11-25 by LucasBrazi06

hasanr68, 2024-10-11:
Hi @ravikarodiya @rccess we are trying to do prepayment. Did you finish this? Would be great to hear what/how you had to do to get this functionality.
Thanks.