sap-labs-france/ev-server #3231: Bug - inactivity status - wrong color

ClaudeROSSI, 2022-01-07:
The inactivity status is suspicious when running automated tests (same meter values / same consumptions, but with a different pricing model).

Here is an example where we get unpredictable colours!

[image]

LucasBrazi06, 2022-01-07:
The color depends on the type of Charging Station.
We didn't put a legend in the UI yet.
Check Utils.getEndOfChargeNotificationIntervalMins()

ClaudeROSSI, 2022-01-07:
good to know!
So, this is not a bug!

p.s.: I was trying to generate a colorful-rainbow inactivity status ;)

LucasBrazi06, 2022-01-07:
> p.s.: I was trying to generate a colorful-rainbow inactivity status ;)

LOL, you will miss some color :D !!!

closed 2022-01-07 by LucasBrazi06