sap-labs-france/ev-server #3252: Enhanced security for OCCP 1.6

guido4096, 2022-01-18:
Curious to know if there already plans to implement "Enhanced security for OCCP 1.6" from here: https://www.openchargealliance.org/uploads/files/OCPP-1.6-security-whitepaper-edition-2.zip ? 

I am thinking that:
- per "Charging Station Registration Token" you would be able to indicate what kind of security the charging station should use once it is registered. Maybe authentication is even useful during 1st time registration with a password that can only be used during 1st time registration.
- The charging station object in Open E-Mobility would get additional flags and a unique password that is to be used by the charging station for authentication. The password would automatically be set on the charging station during the initial registration. 

If not, what is your view on operating your back-end where the charging statations are operating from a regular public internet address space and you cannot set limitations based on ip address?

LucasBrazi06, 2022-01-19:
If I'm not mistaken, this is the retrofit of the Plug & Charge of the ISO15118 standard in OCPP 1.6.
It's not yet planned.
We only accept WSS as connection to our backend, at least the communication is protected with TLS.

guido4096, 2022-01-20:
No, this is not about Plu & Charge. This is just the authentication between a charging station and the backend. It is the retrofit of security parts of OCPP 2.0.1 to 1.6-J. 

If you only rely on the information in the URL to authorize a charging station, it is easy to abuse the backend with a malicious charging system and send false data. 

Short term I believe it should be possible to protect the backend from malicious charging stations by putting a reverse proxy in front of the WSS connection that enforces authentication. 

Long term, especially when working OCCP 2.0.1, I believe this is better integrated in the backend to make management easier.

LucasBrazi06, 2022-01-20:
Thanks for the info!

> If you only rely on the information in the URL to authorize a charging station, it is easy to abuse the backend with a malicious charging system and send false data.

Yes if you have access to the charging station admin console! To mitigate this we put a token in the URL that the customer can change overtime which is check in every OCPP request.

We have currently an AWS Load Balancer for connecting the charging stations and in our new Kubernetes cluster we'll have NGINX.
I would be glad to have you expertise in this area :-).

Yes, I agree, long term, we need this integration!!!
We also experienced that charging station manufacturers are very slow to implement such enhancements.

LucasBrazi06, 2022-01-21:
@guido4096 
Thanks for this precious information!
You are much deeper in security that I will ever be :-).
Let me know, we can check it on our QA system where we have an NGINX ingress.
https://kubernetes.github.io/ingress-nginx/