sap-labs-france/ev-server #3253: Wrong escaping of HTML tags in long texts (in integration settings)

ClaudeROSSI, 2022-01-18:
Tags are not escaped anymore:

Example below tag BR!

[image]

ClaudeROSSI, 2022-01-18:
@LucasBrazi06 : Tu sais d'ou ca peux venir?

LucasBrazi06, 2022-01-18:
Just merged a PR that fixed it ;-)

ClaudeROSSI, 2022-01-21:
We also have a problem with the text of the confirmation dialogs!

[image]

LucasBrazi06, 2022-01-21:
> We also have a problem with the text of the confirmation dialogs!
> 
> [image]

In the i18n we could put an Array of strings, 1 per line and detect in the dialog the multiline text and add the right HTML tags in that case (<br/>.)
What do you think?

ClaudeROSSI, 2022-01-24:
> > We also have a problem with the text of the confirmation dialogs!
> > [image]
> 
> In the i18n we could put an Array of strings, 1 per line and detect in the dialog the multiline text and add the right HTML tags in that case (`<br/>`.) What do you think?

Let me check ... I was using the HTML Tag escaping in the billing as well (when asking to confirm the billing activation). This was working fine at that time. I need to understand why!

ClaudeROSSI, 2022-01-24:
> Let me check ... I was using the HTML Tag escaping in the billing as well (when asking to confirm the billing activation). This was working fine at that time. I need to understand why!

The root cause of the new behavior is a security fix from Siwar to prevent xss injection attacks!

[image]

ClaudeROSSI, 2022-01-24:
> The root cause of the new behavior is a security fix from Siwar to prevent xss injection attacks!
>
> [image]

@LucasBrazi06 : We have the same security issue with the EULA text. I would therefore suggest reverting the fix from Siwar (for now). 
What do you think?

Here is the previous behavior!
[image]

LucasBrazi06, 2022-01-24:
> @LucasBrazi06 : We have the same security issue with the EULA text. I would therefore suggest reverting the fix from Siwar (for now).
> What do you think?

I agree as the source of the HTML is not provided from the end-user but by the application itself.
We can think of the multi line in the i18n mid-term to remove the HTML tags in it which is not nice.

closed 2022-01-26 by LucasBrazi06