sap-labs-france/ev-server #3373: Concur connection error 400

KinzaBS, 2022-02-22:


KinzaBS, 2022-02-22:
Hi, We have an issue on the concur integration. When a new user try to connect to its concur account, we have an error. 
[image]

KinzaBS, 2022-02-22:
https://www-emea.api.concursolutions.com/oauth2/v0/authorize%20%20%20%20%20%20%20%20?client_id=6cf707fb-9161-48fa-94fe-9e003be680df%20%20%20%20%20%20%20%20&response_type=code&scope=EXPRPT%20%20%20%20%20%20%20%20&redirect_uri=https://slfcah.e-mobility-group.com/users/connections%20%20%20%20%20%20%20%20&state={%22connector%22:%22concur%22,%22appId%22:%225d023d394284f700062fb70a%22,%22userId%22:%225e17090ad5e1b9000659c1c2%22}

When we remove the %20 of the URL it works.

LucasBrazi06, 2022-02-22:
@RohanAntony Bad eslint fixes broke the URL. Double check others. Thanks

[image]

@KinzaBS Fixed both Concur and Mercedez connections.

closed 2022-02-22 by LucasBrazi06