sap-labs-france/ev-server #3401: ev-sap-charging-station-templates.git not available

lblabr, 2022-03-06:
i'm trying to build the ev-server without success. i can't get the submodule e.g. ev-sap-charging-station-templates. clicking the linked submodul in github does not work

LucasBrazi06, 2022-03-07:
You cannot access sub-modules, it's private.
It should not break the build.
Could you add more details?

lblabr, 2022-03-08:
i used the the docker makefile, make all it did not work, after disable submodules it build worked. 
but after starting the container it reclaims missing baseSecureUrl parameter

in config its set to:

"WSDLEndpoint": {
"baseSecureUrl": "WSDLEndpoint"
},
"JsonEndpoint": {
"baseSecureUrl": "JsonEndpoint"
},

How the parameter should be filled? Could you pls add an preset in the config-template?

thanks a lot!

lblabr, 2022-03-08:
found it, i worked according your description in the README

```
In **src/assets** folder there are two templates already provided named **config-template-http.json** for HTTP and **config-template-https.json** for HTTPS.

Choose one and rename it to **config.json**.

```

but there are already provided config-examples in the docker directory...

lblabr, 2022-03-08:
by docker build an config.json is deployed:

P },
  "WSDLEndpoint": {
    "baseUrl": "http://server:8000"
  },
  "JsonEndpoint": {
    "baseUrl": "ws://server:8010"
  },
  "OCPIEndpoint": {
    "baseUrl": "http://server:9090"
  },
  "AsyncTask": {
    "active": true,
    "nbrTasksInParallel": 4
  },

whats wrong with that ?

closed 2022-03-11 by lblabr

testocpp, 2022-04-06:
Hi mate, 

Did above base URL for Jsonendpoint work?

LucasBrazi06, 2022-04-06:
> Hi mate,
> 
> Did above base URL for Jsonendpoint work?

No you can only use baseSecureUrl instead of baseUrl.

testocpp, 2022-04-06:
> > Hi mate,
> > Did above base URL for Jsonendpoint work?
> 
> No you can only use baseSecureUrl instead of baseUrl.

Thanks for replyin back. Just to be clear do it need to be https and wss ? 

Also does Host in following centralsystems and baseSecureURL for jasonEndpoint need to be consistent? For example is company has url of "abc.com" in that case 
Host "abc.com" and 
 "JsonEndpoint": {
    "baseSecureUrl": "abc.com"
  },

"CentralSystems": [
    {
      "type": "ocpp",
      "implementation": "soap",
      "protocol": "http",
      "host": "localhost",
      "port": 8000,
      "debug": false
    }

Apologies if question is really dumb as I am trying to learn coding and using this as an exercise. 

Regards

LucasBrazi06, 2022-04-06:
No problem ;-).
No you can put ws or http in it but all URLs won't be secured.

testocpp, 2022-04-06:
> No problem ;-).
> No you can put ws or http in it but all URLs won't be secured.

Thanks again. I forgot to include http and ws. Is port number also required under the jsonendpoint ?

LucasBrazi06, 2022-04-06:
Nope, not mandatory.