sap-labs-france/ev-server #3402: error on startup: must have required property baseSecureUrl

lblabr, 2022-03-06:
with submodules=false and "make All" i do get:

```
2022-03-06T09:18:29: PM2 log: App [start:0] exited with code [1] via signal [SIGINT]
2022-03-06T09:18:29: PM2 log: App [start:0] starting in -fork mode-
2022-03-06T09:18:29: PM2 log: App [start:0] online
AppError: Property '/WSDLEndpoint': must have required property 'baseSecureUrl', Property '/JsonEndpoint': must have required property 'baseSecureUrl'
    at ConfigurationValidatorStorage.validate (/usr/app/dist/webpack:/ev-server/src/validator/SchemaValidator.ts:121:13)
    at ConfigurationValidatorStorage.validateConfiguration (/usr/app/dist/webpack:/ev-server/src/storage/mongodb/validator/ConfigurationValidatorStorage.ts:30:17)
    at Function.getConfig (/usr/app/dist/webpack:/ev-server/src/utils/Configuration.ts:298:74)
    at Function.getNotificationConfig (/usr/app/dist/webpack:/ev-server/src/utils/Configuration.ts:86:40)
    at Object.3305 (/usr/app/dist/webpack:/ev-server/src/notification/NotificationHandler.ts:22:53)
    at __webpack_require__ (/usr/app/dist/webpack:/ev-server/webpack/bootstrap:19:1)
    at Object.516 (/usr/app/dist/webpack:/ev-server/src/integration/billing/BillingIntegration.ts:15:1)
    at __webpack_require__ (/usr/app/dist/webpack:/ev-server/webpack/bootstrap:19:1)
    at Object.8246 (/usr/app/dist/webpack:/ev-server/src/integration/billing/stripe/StripeBillingIntegration.ts:13:1)
    at __webpack_require__ (/usr/app/dist/webpack:/ev-server/webpack/bootstrap:19:1)
    at Object.9341 (/usr/app/dist/webpack:/ev-server/src/integration/billing/BillingFactory.ts:8:1)
    at __webpack_require__ (/usr/app/dist/webpack:/ev-server/webpack/bootstrap:19:1)
    at Object.1395 (/usr/app/dist/webpack:/ev-server/src/async-task/tasks/BillTransactionAsyncTask.ts:4:1)
    at __webpack_require__ (/usr/app/dist/webpack:/ev-server/webpack/bootstrap:19:1)
    at Object.7654 (/usr/app/dist/webpack:/ev-server/src/async-task/AsyncTaskManager.ts:7:1)
    at __webpack_require__ (/usr/app/dist/webpack:/ev-server/webpack/bootstrap:19:1) {
  params: {
    errorCode: 500,
    message: "Property '/WSDLEndpoint': must have required property 'baseSecureUrl', Property '/JsonEndpoint': must have required property 'baseSecureUrl'",
    module: 'ConfigurationValidatorStorage',
    method: 'validate',
    detailedMessages: { errors: [Array], data: [Object], schema: [Object] }
  }
}
```

any hints ?

LucasBrazi06, 2022-03-07:
You are missing property JsonEndpoint in config.json ;-).

closed 2022-03-07 by LucasBrazi06

lblabr, 2022-03-08:
do you have an example? 
for testing the error i do have:

 "WSDLEndpoint": {
    "baseSecureUrl": "WSDLEndpoint"
  },
  "JsonEndpoint": {
    "baseSecureUrl": "JsonEndpoint"
  },

How the parameter should be filled?

lblabr, 2022-03-08:
with docker that cofiguration is distrubuted, whats wrong with that?

 },
  "WSDLEndpoint": {
    "baseUrl": "http://server:8000"
  },
  "JsonEndpoint": {
    "baseUrl": "ws://server:8010"
  },
  "OCPIEndpoint": {
    "baseUrl": "http://server:9090"
  },
  "AsyncTask": {
    "active": true,
    "nbrTasksInParallel": 4
  },