sap-labs-france/ev-server #3405: I am trying to run this on Node 16.14.0, NPM 8.3.1 and MongoDB 4.4.13, but I am facing the same error.

dignitech-rakesh, 2022-03-09:
I am trying to run this on Node 16.14.0, NPM 8.3.1 and MongoDB 4.4.13, but I am facing the same error.
[image]
[image]

Please advise.

_Originally posted by @dignitech-rakesh in https://github.com/sap-labs-france/ev-server/issues/3397#issuecomment-1061822991_

LucasBrazi06, 2022-03-09:
Dups of #3397 ?

dignitech-rakesh, 2022-03-09:
> Dups of #3397 ?

Yes, but I am curious to know that will it specifically work on npm 8.5.2 version?

LucasBrazi06, 2022-03-09:
You are using 8.3.1 here, not 8.5.2

closed 2022-03-09 by dignitech-rakesh