sap-labs-france/ev-server #3407: TS2615: Type of property 'siteAreas' circularly references itself

nazlikrmn, 2022-03-09:
Types are cyclically calling each other.
E.g:
Transaction-> site?: Site-> siteAreas?: SiteArea[]-> site: Site; 

`TS2615: Type of property 'site' circularly references itself in mapped type '{ id: ["id"]; name: ["name"]; address: ["address", "coordinates", number] | ["address", "address1"] | ["address", "address2"] | ["address", "postalCode"] | ["address", "city"] | [...] | [...] | [...]; ... 33 more ...; canReadChargingStations: [...]; }'.
 @ ./src/async-task/tasks/BillTransactionAsyncTask.ts 13:53-104
 @ ./src/async-task/AsyncTaskManager.ts 7:59-102
 @ ./src/start.ts 6:51-91`

`TS2615: Type of property 'siteArea' circularly references itself in mapped type '{ id: ["id"]; createdBy: ["createdBy"]; createdOn: ["createdOn"]; lastChangedBy: ["lastChangedBy"]; lastChangedOn: ["lastChangedOn"]; coordinates: ["coordinates", number]; issuer: ["issuer"]; ... 56 more ...; oicpData: [...] | ... 59 more ... | [...]; }'.
 @ ./src/async-task/tasks/BillTransactionAsyncTask.ts 13:53-104
 @ ./src/async-task/AsyncTaskManager.ts 7:59-102
 @ ./src/start.ts 6:51-91
`

guido4096, 2022-03-23:
I ran into the same. I believe it is related to changes in mongodb. I can detail the exact version that causes the problem and maybe fix the code. If ok, please assign this issue to me.

LucasBrazi06, 2022-03-23:
When do you have this error?
Could you also provide Node/Npm/MongoDB version?

guido4096, 2022-03-23:
yes, will do but can only do this evening

guido4096, 2022-03-23:
One type of change that allowed to get the compile through is by the changing the type Transaction in below line to any . Such change is needed in multiple files. I don't know TypeScript good enough to estimate if this is a good fix though. 

[image]

guido4096, 2022-03-23:
I did  not manage to test, but pretty sure it is related to mongodb version v4.3.0. This introduced Type inference for nested documents: "... support for key auto-completion and type hinting on nested documents!" . See the documented limitation here:

[image]

More details in the release notes here, scroll to version v4.3.0
https://github.com/mongodb/node-mongodb-native/releases

guido4096, 2022-03-23:
Have a look at this commit to find where the error all occurs: https://github.com/guido4096/ev-server/commit/9cbd2deeebe902e221c762f086f4125cec6c6730

nazlikrmn, 2022-03-24:
I can share more information on the subject. 
I'm working on v2.6.2 tag. 

> {
  'ev-server': '2.6.2',
  npm: '8.3.1',
  node: '16.14.0',
  v8: '9.4.146.24-node.20',
  uv: '1.43.0',
  zlib: '1.2.11',
  brotli: '1.0.9',
  ares: '1.18.1',
  modules: '93',
  nghttp2: '1.45.1',
  napi: '8',
  llhttp: '6.0.4',
  openssl: '1.1.1m+quic',
  cldr: '40.0',
  icu: '70.1',
  tz: '2021a3',
  unicode: '14.0',
  ngtcp2: '0.1.0-DEV',
  nghttp3: '0.1.0-DEV'
}

I have  "mongodb": "^4.2.2", on package.json file. And I am connecting to mongodb atlas .I would like to point out that I am not working with a local mongodb. 

I used any instead of SiteArea and Site on src/types/ChargingStation.ts and src/types/SiteArea.ts to solve the problem and continue using home-server. 

Since I don't think this is a good enough solution, I felt the need to open an issue, thinking that you are more familiar with home-server architecture. If you have a diagram that will help me develop a better solution, if you can share it, I would like to support it.

guido4096, 2022-03-24:
The ^ in the line "mongodb": "^4.2.2" says that version 4.2.2 or higher is allowed. Please run npm list mongodb in the ev-server directory to find the actual version that is used. This is likely 4.3.0 or higher.

LucasBrazi06, 2022-03-24:
> npm install mongodb@4.3.0
> and then a
> npm run build:prod
> => npm list mongodb

Use MongoDB 4.4

LucasBrazi06, 2022-03-24:
My version:

```
{
  'ev-dashboard': '2.6.3',
  npm: '8.1.0',
  node: '16.13.0',
  v8: '9.4.146.19-node.13',
  uv: '1.42.0',
  zlib: '1.2.11',
  brotli: '1.0.9',
  ares: '1.17.2',
  modules: '93',
  nghttp2: '1.45.1',
  napi: '8',
  llhttp: '6.0.4',
  openssl: '1.1.1l+quic',
  cldr: '39.0',
  icu: '69.1',
  tz: '2021a',
  unicode: '13.0',
  ngtcp2: '0.1.0-DEV',
  nghttp3: '0.1.0-DEV'
}
```

guido4096, 2022-03-24:
mongodb 4.4.13 does not exit. Well, that is not 100% accurate. MongoDB as the actual database server does exist in version 4.4.13:
https://www.mongodb.com/docs/upcoming/release-notes/4.4/

But this error is not about the database itself. It is about the MongoDB Node.js package which contains the javascript/typescript  client interface. https://github.com/mongodb/node-mongodb-native/releases

When using version 4.3.0 or higher, things break.

LucasBrazi06, 2022-03-24:
I have that in my console for the server when I pull the MongoDB image:
[image]

For the MongoDB driver, I haven't tested the V4.3 but we can fix it and use 4.2.x only in the package.json.

guido4096, 2022-03-24:
Limiting the MongdoDB driver to 4.2.2 will fix it short term. But it blocks taking updates of this driver, which might become a problem in the future. 

P.S. In my own setup I also have MongoDB server 4.4.x; and that also works well combined with the driver version 4.2.2

LucasBrazi06, 2022-03-24:
Yes I agree, but for the time being this is not a prio for the team.

LucasBrazi06, 2022-03-24:
I just tested it and reproduced the issue with the driver 4.4.1 and I maybe found a simple solution to implement.
I'm working on it and let you know.

LucasBrazi06, 2022-03-25:
I created a branch for migrating to MDB driver 4.4: [migration-mongodb-driver-4.4](https://github.com/sap-labs-france/ev-server/tree/migration-mongodb-driver-4.4)

We still have Unit Tests issues with this error:
`MongoServerError: Unsupported conversion from object to string in $convert with no onError value`

nazlikrmn, 2022-03-28:
this is sufficient for my use-case.Thanks for the solution. 
`3/28/2022, 3:02:23 PM - Rest, Json, Ocpi, Oicp, OData server has been started successfuly in 25.473 secs`

closed 2022-03-29 by nazlikrmn