sap-labs-france/ev-server #3411: Could somebody provide a working config.json fro Docker environment ?

lblabr, 2022-03-10:
`docker-compose-wait - Everything's fine, the application can now start!,
--------------------------------------------------------,
2022-03-09T11:25:46: PM2 log: Launching in no daemon mode,
2022-03-09T11:25:47: PM2 log: App [start:0] starting in -fork mode-,
2022-03-09T11:25:47: PM2 log: App [start:0] online,
[Warning] Missing property 'Firebase' in config.json,
3/9/2022, 11:26:00 AM - e-Mobility Server is starting...,
3/9/2022, 11:26:00 AM - NodeJS is started in 'production' mode,
[Warning] Missing property 'OICPService' in config.json,
[Warning] Missing property 'Monitoring' in config.json,
3/9/2022, 11:26:00 AM - Connecting to the Database...,
3/9/2022, 11:26:00 AM - Connecting to 'mongodb'...,
3/9/2022, 11:26:00 AM - TypeError: Cannot read properties of undefined (reading 'toString'),
--------------------------------------------------------,
 docker-compose-wait 2.7.3,
---------------------------,
Starting with configuration:,
 - Hosts to be waiting for: [ev_mongo:27017],
 - Timeout before failure: 90 seconds ,
 - TCP connection timeout before retry: 5 seconds ,
 - Sleeping time before checking for hosts availability: 0 seconds,
 - Sleeping time once all hosts are available: 3 seconds,
 - Sleeping time between retries: 1 seconds,
--------------------------------------------------------,
Checking availability of ev_mongo:27017,
Host ev_mongo:27017 is now available!,
--------------------------------------------------------,
Waiting 3 seconds after hosts availability,
--------------------------------------------------------,
docker-compose-wait - Everything's fine, the application can now start!,
--------------------------------------------------------,
2022-03-10T06:36:38: PM2 log: Launching in no daemon mode,
2022-03-10T06:36:39: PM2 log: App [start:0] starting in -fork mode-,
2022-03-10T06:36:39: PM2 log: App [start:0] online,
[Warning] Missing property 'Firebase' in config.json,
3/10/2022, 6:37:16 AM - e-Mobility Server is starting...,
3/10/2022, 6:37:16 AM - NodeJS is started in 'production' mode,
[Warning] Missing property 'OICPService' in config.json,
[Warning] Missing property 'Monitoring' in config.json,
3/10/2022, 6:37:16 AM - Connecting to the Database...,
3/10/2022, 6:37:16 AM - Connecting to 'mongodb'...,
3/10/2022, 6:37:16 AM - TypeError: Cannot read properties of undefined (reading 'toString'),
--------------------------------------------------------,
 docker-compose-wait 2.7.3,
---------------------------,
Starting with configuration:,
 - Hosts to be waiting for: [ev_mongo:27017],
 - Timeout before failure: 90 seconds ,
 - TCP connection timeout before retry: 5 seconds ,
 - Sleeping time before checking for hosts availability: 0 seconds,
 - Sleeping time once all hosts are available: 3 seconds,
 - Sleeping time between retries: 1 seconds,
--------------------------------------------------------,
Checking availability of ev_mongo:27017,
Host ev_mongo:27017 is now available!,
--------------------------------------------------------,
Waiting 3 seconds after hosts availability,
--------------------------------------------------------,
docker-compose-wait - Everything's fine, the application can now start!,
--------------------------------------------------------,
2022-03-10T06:46:13: PM2 log: Launching in no daemon mode,
2022-03-10T06:46:13: PM2 log: App [start:0] starting in -fork mode-,
2022-03-10T06:46:13: PM2 log: App [start:0] online,
[Warning] Missing property 'Firebase' in config.json,
3/10/2022, 6:46:31 AM - e-Mobility Server is starting...,
3/10/2022, 6:46:31 AM - NodeJS is started in 'production' mode,
[Warning] Missing property 'OICPService' in config.json,
[Warning] Missing property 'Monitoring' in config.json,
3/10/2022, 6:46:31 AM - TypeError: Cannot read properties of undefined (reading 'toString'),
3/10/2022, 6:46:31 AM - Connecting to the Database...,
3/10/2022, 6:46:31 AM - Connecting to 'mongodb'...,
--------------------------------------------------------,
 docker-compose-wait 2.7.3,
---------------------------,
Starting with configuration:,
 - Hosts to be waiting for: [ev_mongo:27017],
 - Timeout before failure: 90 seconds ,
 - TCP connection timeout before retry: 5 seconds ,
 - Sleeping time before checking for hosts availability: 0 seconds,
 - Sleeping time once all hosts are available: 3 seconds,
 - Sleeping time between retries: 1 seconds,
--------------------------------------------------------,
Checking availability of ev_mongo:27017,
Host ev_mongo:27017 is now available!,
--------------------------------------------------------,
Waiting 3 seconds after hosts availability,
--------------------------------------------------------,
docker-compose-wait - Everything's fine, the application can now start!,
--------------------------------------------------------,
2022-03-10T06:56:36: PM2 log: Launching in no daemon mode,
2022-03-10T06:56:37: PM2 log: App [start:0] starting in -fork mode-,
2022-03-10T06:56:37: PM2 log: App [start:0] online,
[Warning] Missing property 'Firebase' in config.json,
3/10/2022, 6:56:51 AM - e-Mobility Server is starting...,
3/10/2022, 6:56:52 AM - NodeJS is started in 'production' mode,
[Warning] Missing property 'OICPService' in config.json,
[Warning] Missing property 'Monitoring' in config.json,
3/10/2022, 6:56:52 AM - Connecting to the Database...,
3/10/2022, 6:56:52 AM - Connecting to 'mongodb'...,
3/10/2022, 6:56:52 AM - TypeError: Cannot read properties of undefined (reading 'toString')`

execution stops more or less after Connecting mongodb

```
root@31f068b1ac5b:/usr/app# ps aux
USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root         1  0.0  0.0   2392   760 ?        Ss   06:56   0:00 /bin/sh -c /wait && pm2-runtime dist/start.js
root         9  1.2  0.3 628804 50380 ?        Sl   06:56   0:01 node /usr/local/bin/pm2-runtime dist/start.js
root        20  9.0  0.9 775760 159772 ?       Ssl  06:56   0:13 node /usr/app/dist/start.js
root        36  3.5  0.0   3872  3244 pts/0    Ss   06:59   0:00 bash
root        43  0.0  0.0   7644  2736 pts/0    R+   06:59   0:00 ps aux
root@31f068b1ac5b:/usr/app# 
```

thanks in advance

lblabr, 2022-03-10:
it seems that it stops somewhere here:

```
Logging.logConsoleDebug(`Connecting to '${this.dbConfig.implementation}'...`);
    // Build EVSE URL
    let mongoUrl: string;
    // URI provided?
    if (this.dbConfig.uri) {
      // Yes: use it
      mongoUrl = this.dbConfig.uri;
    // Build URI without replicaset
    } else {
      const uri: MongoUriConfig = {
        host: urlencode(this.dbConfig.host),
        port: Utils.convertToInt(urlencode(this.dbConfig.port.toString())),
        username: urlencode(this.dbConfig.user),
        password: urlencode(this.dbConfig.password),
        database: urlencode(this.dbConfig.database),
        options: {
          readPreference: this.dbConfig.readPreference ? this.dbConfig.readPreference as ReadPreferenceMode : ReadPreferenceMode.secondaryPreferred
        }
      };
      // Set the Replica Set
      if (this.dbConfig.replicaSet) {
        uri.options.replicaSet = this.dbConfig.replicaSet;
      }
      mongoUrl = mongoUriBuilder(uri);
    }
    // Connect to EVSE
    Logging.logConsoleDebug(`Connecting to '${mongoUrl}'`);
```

could somebody provide a valid uri for docker setup?

the error:
`3/10/2022, 2:33:40 PM - TypeError: Cannot read properties of undefined (reading 'toString')
`
is triggerd from 
`if (this.dbConfig.uri)
`
?
by uri in config is set to "null" ??

beraterloesung, 2022-03-18:
you have to set proper host and port - if you run mongo and mail based on the docker-compose-local-env.yml, then it works with:
...
    "host": "localhost",
    "port": 27017,
...

LucasBrazi06, 2022-05-09:
It's up to you to provide your config.json according the provided template in asset directory.
This config is validated against a schema.

closed 2022-05-09 by LucasBrazi06