sap-labs-france/ev-server #3413: pls provide an example for an valid chargingstation-template

lblabr, 2022-03-12:
pls provide an valid example for an chargingstation-template for an specific vendor, i would like to adapt for different wallboxes

thanks in advance

lblabr, 2022-03-12:
any hints for my first try to adapt webasto:

```
{
	"id": "LB1000",
    "chargePointVendor": "Webasto",
    "extraFilters": {
        "chargePointModel": "Webasto_AC_Live"
	},
    "hash": "hash",
    "hashTechnical": "hashTechnical",
    "hashCapabilities": "hashCapabilities",
    "hashOcppStandard": "1.6",
    "hashOcppVendor": "hashOcppVendor",
    "technical": {
        "maximumPower": 22,
        "voltage": 230,
        "powerLimitUnit": "kW",
        "chargePoints": [
			{
				"chargePointID": 1,
				"currentType": "AC",
				"amperage": 32,
				"numberOfConnectedPhase": 3,
				"cannotChargeInParallel": false,
				"sharePowerToAllConnectors": false,
				"ocppParamForPowerLimitation": "DlmOperatorPhase1Limit_dlm",
				"power": 22080,
				"connectorIDs": [1]
            }
		],
        "connectors": [
			{
				"connectorId": 1,
				"type": "T2",
				"power": 22080,
				"chargePointID": 1
            }
		]
    },
    "capabilities": [
		{	"supportedFirmwareVersions": ["5.00-6542", "5.11.0-10125", "5.11.1-10129"],
			"supportedOcppVersions": ["1.5","1.6"],
			"capabilities": {
					"supportStaticLimitation": true,
					"supportChargingProfiles": true,
					"supportRemoteStartStopTransaction": true,
					"supportUnlockConnector": false,
					"supportReservation": false,
					"supportCreditCard": false,
					"supportRFIDCard": true
			}
		}
	],
    "ocppStandardParameters": [
		{
			"supportedFirmwareVersions": ["5.00-6542","5.11.0-10125","5.11.1-10129"],
			"supportedOcppVersions": ["1.5","1.6"],
			"parameters": {
              "AuthorizationCacheEnabled": "0",
              "AuthorizeRemoteTxRequests": "1",
              "ClockAlignedDataInterval": "0",
              "ConnectionTimeOut": "120",
              "LocalAuthorizeOffline": "1",
              "LocalPreAuthorize": "1",
              "MeterValuesSampledData": "Energy.Active.Import.Register,Power.Active.Import,Current.Import,Voltage",
              "MeterValueSampleInterval": "60",
              "StopTransactionOnEVSideDisconnect": "1",
              "StopTransactionOnInvalidId": "1",
              "UnlockConnectorOnEVSideDisconnect": "1",
              "WebSocketPingInterval": "60",
            },

        }
	],
    "ocppVendorParameters": [
		{
			"supportedFirmwareVersions": ["5.00-6542","5.11.0-10125","5.11.1-10129"],
			"supportedOcppVersions": ["1.5","1.6"],
			"parameters": {
				"xyz": "xyz"
            }
        }
    ]
}
```

LucasBrazi06, 2022-03-13:
This is consulting service, I'm sorry.
It has to pass the Json schema here: 
`src/assets/schemas/chargingstation/chargingstation-template.json`

And you can put you json file in this path:
`src/assets/charging-station-templates/charging-stations.json`

charging-stations.json file is an array of template so put your Json in [ ].
At start-up this file will be read and stored in the DB to be used during Boot Notification.

closed 2022-03-13 by LucasBrazi06

guido4096, 2022-04-08:
this works quite well, just had two typos with comma's. Fixed below version. That at least gets the format right, but you still need to make sure you get all information correct.

{
	"id": "LB1000",
    "chargePointVendor": "Webasto",
    "extraFilters": {
        "chargePointModel": "Webasto_AC_Live"
	},
    "hash": "hash",
    "hashTechnical": "hashTechnical",
    "hashCapabilities": "hashCapabilities",
    "hashOcppStandard": "1.6",
    "hashOcppVendor": "hashOcppVendor",
    "technical": {
        "maximumPower": 22,
        "voltage": 230,
        "powerLimitUnit": "kW",
        "chargePoints": [
			{
				"chargePointID": 1,
				"currentType": "AC",
				"amperage": 32,
				"numberOfConnectedPhase": 3,
				"cannotChargeInParallel": false,
				"sharePowerToAllConnectors": false,
				"ocppParamForPowerLimitation": "DlmOperatorPhase1Limit_dlm",
				"power": 22080,
				"connectorIDs": [1]
            }
		],
        "connectors": [
			{
				"connectorId": 1,
				"type": "T2",
				"power": 22080,
				"chargePointID": 1
            }
		]
    },
    "capabilities": [
		{	"supportedFirmwareVersions": ["5.00-6542", "5.11.0-10125", "5.11.1-10129"],
			"supportedOcppVersions": ["1.5","1.6"],
			"capabilities": {
					"supportStaticLimitation": true,
					"supportChargingProfiles": true,
					"supportRemoteStartStopTransaction": true,
					"supportUnlockConnector": false,
					"supportReservation": false,
					"supportCreditCard": false,
					"supportRFIDCard": true
			}
		}
	],
    "ocppStandardParameters": [
		{
			"supportedFirmwareVersions": ["5.00-6542","5.11.0-10125","5.11.1-10129"],
			"supportedOcppVersions": ["1.5","1.6"],
			"parameters": {
              "AuthorizationCacheEnabled": "0",
              "AuthorizeRemoteTxRequests": "1",
              "ClockAlignedDataInterval": "0",
              "ConnectionTimeOut": "120",
              "LocalAuthorizeOffline": "1",
              "LocalPreAuthorize": "1",
              "MeterValuesSampledData": "Energy.Active.Import.Register,Power.Active.Import,Current.Import,Voltage",
              "MeterValueSampleInterval": "60",
              "StopTransactionOnEVSideDisconnect": "1",
              "StopTransactionOnInvalidId": "1",
              "UnlockConnectorOnEVSideDisconnect": "1",
              "WebSocketPingInterval": "60"
            }

        }
	],
    "ocppVendorParameters": [
		{
			"supportedFirmwareVersions": ["5.00-6542","5.11.0-10125","5.11.1-10129"],
			"supportedOcppVersions": ["1.5","1.6"],
			"parameters": {
				"xyz": "xyz"
            }
        }
    ]
}

testocpp, 2022-04-08:
@guido4096  Thanks will try to use this. What is the purpose of this file? Will different brand/product_lines have their own templates under the [ ]? Thanks