sap-labs-france/ev-server #3418: Error when "npm run start"

emry78, 2022-03-16:
Hi All,
Done config file config.json . Then i run "npm run start". and end of this comeout with this error.

- notification handler
- billing integration
- stripe billing integration
- billing factory
- bill transcation async task
- async task manager

how to solve this. need to config other file? need help and advise. Thanks

The error shows as below:-

SyntaxError: Unexpected string in JSON at position 3354
    at JSON.parse (<anonymous>)
    at Function.getConfig (/home/snuc/ev-server/dist/webpack:/ev-server/src/utils/Configuration.ts:294:30)
    at Function.getNotificationConfig (/home/snuc/ev-server/dist/webpack:/ev-server/src/utils/Configuration.ts:86:40)
    at Object.3305 (/home/snuc/ev-server/dist/webpack:/ev-server/src/notification/NotificationHandler.ts:22:53)
    at __webpack_require__ (/home/snuc/ev-server/dist/webpack:/ev-server/webpack/bootstrap:19:1)
    at Object.516 (/home/snuc/ev-server/dist/webpack:/ev-server/src/integration/billing/BillingIntegration.ts:12:1)
    at __webpack_require__ (/home/snuc/ev-server/dist/webpack:/ev-server/webpack/bootstrap:19:1)
    at Object.8246 (/home/snuc/ev-server/dist/webpack:/ev-server/src/integration/billing/stripe/StripeBillingIntegration.ts:13:1)
    at __webpack_require__ (/home/snuc/ev-server/dist/webpack:/ev-server/webpack/bootstrap:19:1)
    at Object.9341 (/home/snuc/ev-server/dist/webpack:/ev-server/src/integration/billing/BillingFactory.ts:8:1)
    at __webpack_require__ (/home/snuc/ev-server/dist/webpack:/ev-server/webpack/bootstrap:19:1)
    at Object.1395 (/home/snuc/ev-server/dist/webpack:/ev-server/src/async-task/tasks/BillTransactionAsyncTask.ts:4:1)
    at __webpack_require__ (/home/snuc/ev-server/dist/webpack:/ev-server/webpack/bootstrap:19:1)
    at Object.7654 (/home/snuc/ev-server/dist/webpack:/ev-server/src/async-task/AsyncTaskManager.ts:7:1)
    at __webpack_require__ (/home/snuc/ev-server/dist/webpack:/ev-server/webpack/bootstrap:19:1)
    at /home/snuc/ev-server/dist/webpack:/ev-server/src/start.ts:5:1

emry78, 2022-03-21:
https://github.com/sap-labs-france/ev-server/issues/3418#issue-1170533977

Any suggestion guys. Thanks

LucasBrazi06, 2022-03-21:
You config.json file is not valid

LucasBrazi06, 2022-03-23:
Check the Json content of this file.

guido4096, 2022-03-23:
I can take this issue to improve the template files to make it accurate, let me know if ok and assign it to me,

guido4096, 2022-03-24:
poolSize attribute was missing from Storage element

closed 2022-03-25 by LucasBrazi06

emry78, 2022-03-25:
> poolSize attribute was missing from Storage element

thank for reply. I will follow the suggestion.

emry78, 2022-03-28:
Sorry when i run again. still got this pblm.

snuc@snuc-NUC11BTMi9:~/ev-server/src/assets$ npm run start

> ev-server@2.6.3 prestart
> npm run build:prod

> ev-server@2.6.3 build:prod
> npm version && webpack --env prod

{
  'ev-server': '2.6.3',
  npm: '8.5.0',
  node: '16.14.1',
  v8: '9.4.146.24-node.20',
  uv: '1.43.0',
  zlib: '1.2.11',
  brotli: '1.0.9',
  ares: '1.18.1',
  modules: '93',
  nghttp2: '1.45.1',
  napi: '8',
  llhttp: '6.0.4',
  openssl: '1.1.1m+quic',
  cldr: '40.0',
  icu: '70.1',
  tz: '2021a3',
  unicode: '14.0',
  ngtcp2: '0.1.0-DEV',
  nghttp3: '0.1.0-DEV'
}
1415 assets
cacheable modules 3.63 MiB 403 modules
modules by path external "moment/locale/ 252 bytes
  external "moment/locale/fr" 42 bytes [built] [code generated]
  external "moment/locale/de" 42 bytes [built] [code generated]
  external "moment/locale/es" 42 bytes [built] [code generated]
  external "moment/locale/it" 42 bytes [built] [code generated]
  external "moment/locale/en-gb" 42 bytes [built] [code generated]
  external "moment/locale/pt-br" 42 bytes [built] [code generated]
modules by path *.js" 84 bytes
  external "decimal.js" 42 bytes [built] [code generated]
  external "uWebSockets.js" 42 bytes [built] [code generated]
+ 69 modules
webpack 5.67.0 compiled successfully in 11933 ms

> ev-server@2.6.3 start
> npm version && npm run start:prod

{
  'ev-server': '2.6.3',
  npm: '8.5.0',
  node: '16.14.1',
  v8: '9.4.146.24-node.20',
  uv: '1.43.0',
  zlib: '1.2.11',
  brotli: '1.0.9',
  ares: '1.18.1',
  modules: '93',
  nghttp2: '1.45.1',
  napi: '8',
  llhttp: '6.0.4',
  openssl: '1.1.1m+quic',
  cldr: '40.0',
  icu: '70.1',
  tz: '2021a3',
  unicode: '14.0',
  ngtcp2: '0.1.0-DEV',
  nghttp3: '0.1.0-DEV'
}

> ev-server@2.6.3 start:prod
> npm version && cross-env NODE_OPTIONS="--max-old-space-size=4096" NODE_ENV=production node -r source-map-support/register dist/start.js

{
  'ev-server': '2.6.3',
  npm: '8.5.0',
  node: '16.14.1',
  v8: '9.4.146.24-node.20',
  uv: '1.43.0',
  zlib: '1.2.11',
  brotli: '1.0.9',
  ares: '1.18.1',
  modules: '93',
  nghttp2: '1.45.1',
  napi: '8',
  llhttp: '6.0.4',
  openssl: '1.1.1m+quic',
  cldr: '40.0',
  icu: '70.1',
  tz: '2021a3',
  unicode: '14.0',
  ngtcp2: '0.1.0-DEV',
  nghttp3: '0.1.0-DEV'
}

/home/snuc/ev-server/dist/webpack:/ev-server/src/utils/Configuration.ts:294
        configuration = JSON.parse(
                             ^
SyntaxError: Unexpected token / in JSON at position 585
    at JSON.parse (<anonymous>)
    at Function.getConfig (/home/snuc/ev-server/dist/webpack:/ev-server/src/utils/Configuration.ts:294:30)
    at Function.getNotificationConfig (/home/snuc/ev-server/dist/webpack:/ev-server/src/utils/Configuration.ts:86:40)
    at Object.3305 (/home/snuc/ev-server/dist/webpack:/ev-server/src/notification/NotificationHandler.ts:22:53)
    at __webpack_require__ (/home/snuc/ev-server/dist/webpack:/ev-server/webpack/bootstrap:19:1)
    at Object.516 (/home/snuc/ev-server/dist/webpack:/ev-server/src/integration/billing/BillingIntegration.ts:16:1)
    at __webpack_require__ (/home/snuc/ev-server/dist/webpack:/ev-server/webpack/bootstrap:19:1)
    at Object.8246 (/home/snuc/ev-server/dist/webpack:/ev-server/src/integration/billing/stripe/StripeBillingIntegration.ts:13:1)
    at __webpack_require__ (/home/snuc/ev-server/dist/webpack:/ev-server/webpack/bootstrap:19:1)
    at Object.9341 (/home/snuc/ev-server/dist/webpack:/ev-server/src/integration/billing/BillingFactory.ts:8:1)
    at __webpack_require__ (/home/snuc/ev-server/dist/webpack:/ev-server/webpack/bootstrap:19:1)
    at Object.1395 (/home/snuc/ev-server/dist/webpack:/ev-server/src/async-task/tasks/BillTransactionAsyncTask.ts:4:1)
    at __webpack_require__ (/home/snuc/ev-server/dist/webpack:/ev-server/webpack/bootstrap:19:1)
    at Object.7654 (/home/snuc/ev-server/dist/webpack:/ev-server/src/async-task/AsyncTaskManager.ts:7:1)
    at __webpack_require__ (/home/snuc/ev-server/dist/webpack:/ev-server/webpack/bootstrap:19:1)
    at /home/snuc/ev-server/dist/webpack:/ev-server/src/start.ts:5:1

LucasBrazi06, 2022-03-28:
This is no longer a problem on my side.
You're config.json is malformed.
Take the time to understand the error!!!

emry78, 2022-04-04:
Thanks for replying. 1 more thing i want to ask. 
after i run the i got this. what means 'missing property 'Monitoring' in config.json'.
Thanks

4/4/2022, 11:33:09 AM - e-Mobility Server is starting...
4/4/2022, 11:33:09 AM - NodeJS is started in 'production' mode
[Warning] Missing property 'Monitoring' in config.json
4/4/2022, 11:33:09 AM - Connecting to the Database...
4/4/2022, 11:33:09 AM - Connecting to 'mongodb'...
4/4/2022, 11:33:09 AM - Connecting to 'mongodb://evse-user:user123@192.168.8.106:27017/evse'
4/4/2022, 11:33:09 AM - Connected to 'mongodb' successfully
4/4/2022, 11:33:09 AM - Connected to the Database successfully in 0.05 secs
4/4/2022, 11:33:09 AM - Migration is starting...
4/4/2022, 11:33:09 AM - Migration has been run successfully in 0.008 secs
4/4/2022, 11:33:09 AM - Starting Rest Server...
4/4/2022, 11:33:09 AM - Starting Soap Server...
4/4/2022, 11:33:09 AM - Starting Json Server...
4/4/2022, 11:33:09 AM - Json Server listening on 'http://192.168.8.106:8010'
4/4/2022, 11:33:09 AM - Starting Ocpi Server...
4/4/2022, 11:33:09 AM - Starting Oicp Server...
4/4/2022, 11:33:09 AM - Starting OData Server...
4/4/2022, 11:33:09 AM - Async Task manager is starting...
4/4/2022, 11:33:09 AM - Rest Server listening on 'http://192.168.8.106:8080'
4/4/2022, 11:33:09 AM - Soap Server listening on 'http://192.168.8.106:8000'
4/4/2022, 11:33:09 AM - Ocpi Server listening on 'http://192.168.8.106:9090'
4/4/2022, 11:33:09 AM - Oicp Server listening on 'http://192.168.8.106:9080'
4/4/2022, 11:33:09 AM - OData Server listening on 'http://192.168.8.106:9292'
4/4/2022, 11:33:09 AM - Async Task manager has been started successfully in 0.054 secs
4/4/2022, 11:33:09 AM - Charging Station templates is being updated...
4/4/2022, 11:33:09 AM - Charging Station templates have been updated successfully in 0.344 secs
4/4/2022, 11:33:09 AM - Local car catalogs are being imported...
4/4/2022, 11:33:09 AM - Local car catalogs has been imported successfully in 0.014 secs
4/4/2022, 11:33:09 AM - Rest, Soap, Json, Ocpi, Oicp, OData server has been started successfuly in 0.503 secs

LucasBrazi06, 2022-04-04:
It's just a warning to expose the monitoring port for Prometheus.