sap-labs-france/ev-server #3442: EV-simulator not working only waiting

mytrkr0, 2022-03-30:
Hi Mr.Serge 

i started ev-simulator npm start and it is started.
[image]

but dashboard is not updating.

[image]

my config.json file is 

```json
{
  "supervisionURLs": [
    "ws://localhost:8010/OCPP16/623c52f5341fffb3b6ed2310"
  ],
  "distributeStationsToTenantsEqually": true,
  "performanceStorage": {
    "enabled": true
  },
  "chargingStationsPerWorker": 1,
  "workerProcess": "workerSet",
  "workerPoolMinSize": 4,
  "workerPoolMaxSize": 16,
  "stationTemplateURLs": [
    {
      "file": "siemens.station-template.json",
      "numberOfStations": 1
    },
    {
      "file": "keba.station-template.json",
      "numberOfStations": 2
    },
    {
      "file": "abb.station-template.json",
      "numberOfStations": 2
    },
    {
      "file": "evlink.station-template.json",
      "numberOfStations": 4
    },
    {
      "file": "schneider.station-template.json",
      "numberOfStations": 1
    }
  ],
  "logStatisticsInterval": 0,
  "logFile": "combined.log",
  "logErrorFile": "error.log"
}
```

and Central Systems config is

```json
  "CentralSystems": [
    {

      "type" : "ocpp",
       "implementation": "json",
      "protocol": "ws",
      "host": "localhost",
      "port": 8010,
      "debug": false
    }
  ]
```

by the way what is supervisionURLs in config.json and how can i fill it so can you type  like #3048 

[image]

thank you for your answer

mytrkr0, 2022-03-30:
[image]

nazlikrmn, 2022-03-30:
If you haven't made any changes on the connection procedure, I think that the url you entered in the supervision URLs might be the problem. example pattern: 
OCPP16/tenantid/tokenid/chargingstation
Did you try change that?

mytrkr0, 2022-03-30:
tenant and organisation are they equal ?

closed 2022-04-20 by mytrkr0

Zakichanu, 2022-09-30:
Hi, can you tell me how did you fix this?

Zakichanu, 2022-09-30:
Just saw #3048 I understand the problem