sap-labs-france/ev-server #3444: Error when starting server "No Rest Web Socket connection to ping"

AstralisSomnium, 2022-03-31:
### Details

I am in the process of setting up my own ev-server on a self hosted Windows machine and I get an error when I want to use the REST API.

I followed the guideline and when I start the server it looks alright, see "server console output" below.
+ config here: [config.zip](https://github.com/sap-labs-france/ev-server/files/8393428/config.zip)
 + All values with "hidden" I deleted them for uploading it here to the public 
+ I added manually to the mongo db the required user inside the collection `default.users`
+ MongoDb log file: [log.zip](https://github.com/sap-labs-france/ev-server/files/8393455/log.zip)

### SignIn error: HTTP 550

I try with Postman to make a `SignIn` call but I get always following error:
[image]

### Server console output on startup

```
3/31/2022, 10:21:26 PM - e-Mobility Server is starting...
3/31/2022, 10:21:26 PM - NodeJS is started in 'production' mode
Missing property 'Monitoring' in config.json
3/31/2022, 10:21:26 PM - Connecting to the Database...
3/31/2022, 10:21:26 PM - Connecting to 'mongodb'...
3/31/2022, 10:21:26 PM - Connecting to 'mongodb://evse-admin:HIDDEN@HIDDEN:27017/evse?authSource=admin'
3/31/2022, 10:21:27 PM - Connected to 'mongodb' successfully
3/31/2022, 10:21:27 PM - Connected to the Database successfully in 0.533 secs
3/31/2022, 10:21:27 PM - Migration is starting...
3/31/2022, 10:21:27 PM - Migration has been run successfully in 0.144 secs
3/31/2022, 10:21:27 PM - Starting Rest Server...
3/31/2022, 10:21:27 PM - Starting Soap Server...
3/31/2022, 10:21:27 PM - Starting Json Server...
3/31/2022, 10:21:27 PM - Json Server listening on 'http://localhost:8010'
3/31/2022, 10:21:27 PM - Starting Ocpi Server...
3/31/2022, 10:21:27 PM - Starting Oicp Server...
3/31/2022, 10:21:27 PM - Starting OData Server...
3/31/2022, 10:21:27 PM - Scheduler is starting...
3/31/2022, 10:21:28 PM - Rest Server listening on 'http://127.0.0.1:80'
3/31/2022, 10:21:28 PM - Soap Server listening on 'http://127.0.0.1:8000'
3/31/2022, 10:21:28 PM - Ocpi Server listening on 'http://127.0.0.1:9090'
3/31/2022, 10:21:28 PM - Oicp Server listening on 'http://127.0.0.1:9080'
3/31/2022, 10:21:28 PM - OData Server listening on 'http://127.0.0.1:9292'
3/31/2022, 10:21:28 PM - Scheduler has been started successfully in 0.604 secs
3/31/2022, 10:21:28 PM - Async Task manager is starting...
3/31/2022, 10:21:28 PM - Async Task manager has been started successfully in 0.158 secs
3/31/2022, 10:21:28 PM - Charging Station templates is being updated...
3/31/2022, 10:21:29 PM - Charging Station templates have been updated successfully in 1.249 secs
3/31/2022, 10:21:30 PM - Local car catalogs are being imported...
3/31/2022, 10:21:44 PM - Local car catalogs has been imported successfully in 13.226 secs
3/31/2022, 10:21:45 PM - Rest, Soap, Json, Ocpi, Oicp, OData server has been started successfuly in 18.397 secs
POST /v1/auth/signin - 550 - 5056.690ms - 0.0732421875Kb
3/31/2022, 10:22:27 PM - No Json Web Socket connection to ping
3/31/2022, 10:22:27 PM - No Rest Web Socket connection to ping
```

## Question

What am I missing in order to get the REST API running?

LucasBrazi06, 2022-04-01:
First of all, have you checked the logs (error)?
You can already rename acceptEula value from 'True' to 'true' and set tenant to an empty string ''.
Let me know.

AstralisSomnium, 2022-04-01:
@LucasBrazi06 thanks for your quick reply, I really appreciate since I am stuck for a while.
I sent the MongoDb logs on the issue itself or do you talk about another log file? If so let me know where I can find the configuration and log file on windows. 
Many thanks!!

LucasBrazi06, 2022-04-01:
Glad to hear that you tried to figure this out on your own before opening an issue :-)
I meant the app logs in mongodb (default.logs)

AstralisSomnium, 2022-04-14:
@LucasBrazi06 awesome by checking these default.logs I could go forward:
+ I added the charging template.json
+ I re-added the user in the default.users and tried again the login
+ Old Logs which helped me to find this out: [evseLogs.zip](https://github.com/sap-labs-france/ev-server/files/8491973/evseLogs.zip)

### Next error `User failed to log in, 2 trial(s) remaining`

 I have double-checked the username and password, its correct!

Now, I get the following error: 
```
{"_id":{"$oid":"62588725c0d9a566a0e5372c"},"userID":null,"actionOnUserID":null,"level":"W","chargingStationID":null,"companyID":null,"siteID":null,"siteAreaID":null,"source":"CentralServer","host":"EV-Server-Dashb","timestamp":{"$date":"2022-04-14T20:42:13.724Z"},"module":"UserStorage","method":"getUsers","action":"Performances","message":"UserStorage.getUsers - 1991 ms - Req 1.82 KB - Res ? KB - 0 rec(s): Execution must be < 1000 ms","detailedMessages":"[\n {\n  \"error\": \"Error: Execution must be < 1000 ms\\n    at Function.traceDatabaseRequestEnd (C:\\\\Users\\\\Rahul\\\\Desktop\\\\git\\\\ev-server\\\\dist\\\\webpack:\\\\ev-server\\\\src\\\\utils\\\\Logging.ts:83:23)\\n    at Function.getUsers (C:\\\\Users\\\\Rahul\\\\Desktop\\\\git\\\\ev-server\\\\dist\\\\webpack:\\\\ev-server\\\\src\\\\storage\\\\mongodb\\\\UserStorage.ts:699:19)\\n    at Function.getUserByEmail (C:\\\\Users\\\\Rahul\\\\Desktop\\\\git\\\\ev-server\\\\dist\\\\webpack:\\\\ev-server\\\\src\\\\storage\\\\mongodb\\\\UserStorage.ts:88:21)\\n    at AuthRouter.handleLogIn (C:\\\\Users\\\\Rahul\\\\Desktop\\\\git\\\\ev-server\\\\dist\\\\webpack:\\\\ev-server\\\\src\\\\server\\\\rest\\\\v1\\\\service\\\\AuthService.ts:74:18)\\n    at Function.handleServerAction (C:\\\\Users\\\\Rahul\\\\Desktop\\\\git\\\\ev-server\\\\dist\\\\webpack:\\\\ev-server\\\\src\\\\server\\\\rest\\\\v1\\\\router\\\\RouterUtils.ts:12:7)\\n    at C:\\\\Users\\\\Rahul\\\\Desktop\\\\git\\\\ev-server\\\\dist\\\\webpack:\\\\ev-server\\\\src\\\\server\\\\rest\\\\v1\\\\router\\\\auth\\\\AuthRouter.ts:29:7\"\n }\n]"}
{"_id":{"$oid":"62588728c0d9a566a0e5372e"},"userID":null,"actionOnUserID":null,"level":"W","chargingStationID":null,"companyID":null,"siteID":null,"siteAreaID":null,"source":"CentralServer","host":"EV-Server-Dashb","timestamp":{"$date":"2022-04-14T20:42:16.765Z"},"module":"UserStorage","method":"saveUserPassword","action":"Performances","message":"UserStorage.saveUserPassword - 1010 ms - Req 0.06 KB - Res ? KB - 0 rec(s): Execution must be < 1000 ms","detailedMessages":"[\n {\n  \"error\": \"Error: Execution must be < 1000 ms\\n    at Function.traceDatabaseRequestEnd (C:\\\\Users\\\\Rahul\\\\Desktop\\\\git\\\\ev-server\\\\dist\\\\webpack:\\\\ev-server\\\\src\\\\utils\\\\Logging.ts:83:23)\\n    at Function.saveUserPassword (C:\\\\Users\\\\Rahul\\\\Desktop\\\\git\\\\ev-server\\\\dist\\\\webpack:\\\\ev-server\\\\src\\\\storage\\\\mongodb\\\\UserStorage.ts:365:19)\\n    at Function.userLoginWrongPassword (C:\\\\Users\\\\Rahul\\\\Desktop\\\\git\\\\ev-server\\\\dist\\\\webpack:\\\\ev-server\\\\src\\\\server\\\\rest\\\\v1\\\\service\\\\AuthService.ts:623:7)\\n    at Function.checkUserLogin (C:\\\\Users\\\\Rahul\\\\Desktop\\\\git\\\\ev-server\\\\dist\\\\webpack:\\\\ev-server\\\\src\\\\server\\\\rest\\\\v1\\\\service\\\\AuthService.ts:769:7)\\n    at AuthRouter.handleLogIn (C:\\\\Users\\\\Rahul\\\\Desktop\\\\git\\\\ev-server\\\\dist\\\\webpack:\\\\ev-server\\\\src\\\\server\\\\rest\\\\v1\\\\service\\\\AuthService.ts:117:7)\\n    at Function.handleServerAction (C:\\\\Users\\\\Rahul\\\\Desktop\\\\git\\\\ev-server\\\\dist\\\\webpack:\\\\ev-server\\\\src\\\\server\\\\rest\\\\v1\\\\router\\\\RouterUtils.ts:12:7)\\n    at C:\\\\Users\\\\Rahul\\\\Desktop\\\\git\\\\ev-server\\\\dist\\\\webpack:\\\\ev-server\\\\src\\\\server\\\\rest\\\\v1\\\\router\\\\auth\\\\AuthRouter.ts:29:7\"\n }\n]"}
{"_id":{"$oid":"6258872ac0d9a566a0e53730"},"userID":{"$oid":"625886c49a114cd1a134af87"},"actionOnUserID":null,"level":"E","chargingStationID":null,"companyID":null,"siteID":null,"siteAreaID":null,"source":"CentralServer","host":"EV-Server-Dashb","timestamp":{"$date":"2022-04-14T20:42:18.786Z"},"module":"AuthService","method":"checkUserLogin","action":"Login","message":"User failed to log in, 2 trial(s) remaining","detailedMessages":"[\n {\n  \"stack\": \"Error: User failed to log in, 2 trial(s) remaining\\n    at Function.userLoginWrongPassword (C:\\\\Users\\\\Rahul\\\\Desktop\\\\git\\\\ev-server\\\\dist\\\\webpack:\\\\ev-server\\\\src\\\\server\\\\rest\\\\v1\\\\service\\\\AuthService.ts:624:13)\\n    at Function.checkUserLogin (C:\\\\Users\\\\Rahul\\\\Desktop\\\\git\\\\ev-server\\\\dist\\\\webpack:\\\\ev-server\\\\src\\\\server\\\\rest\\\\v1\\\\service\\\\AuthService.ts:769:7)\\n    at AuthRouter.handleLogIn (C:\\\\Users\\\\Rahul\\\\Desktop\\\\git\\\\ev-server\\\\dist\\\\webpack:\\\\ev-server\\\\src\\\\server\\\\rest\\\\v1\\\\service\\\\AuthService.ts:117:7)\\n    at Function.handleServerAction (C:\\\\Users\\\\Rahul\\\\Desktop\\\\git\\\\ev-server\\\\dist\\\\webpack:\\\\ev-server\\\\src\\\\server\\\\rest\\\\v1\\\\router\\\\RouterUtils.ts:12:7)\\n    at C:\\\\Users\\\\Rahul\\\\Desktop\\\\git\\\\ev-server\\\\dist\\\\webpack:\\\\ev-server\\\\src\\\\server\\\\rest\\\\v1\\\\router\\\\auth\\\\AuthRouter.ts:29:7\"\n }\n]"}
```

Here my post request:
```json
{
    "email": "super.admin@ev.com",
    "password": "test_1234_Rahul",
    "acceptEula": true,
    "tenant": ""
}
```

Here the script how I added my user:
```
[... cut ...]
    sendOfflineChargingStations: false
  },
  phone: null,
  password: 'test_1234_Rahul',
  passwordBlockedUntil: null,
  passwordResetHash: null,
  passwordWrongNbrTrials: NumberInt(0),
  eulaAcceptedHash: 'c308ac57857ce483ef1bb50fe8c1bc2bc3b5fcf067114c8b4a3a7abf9896c45f',
  eulaAcceptedOn: ISODate('2020-04-02T00:00:00.000+0000'),
  eulaAcceptedVersion: 28,
  role: 'S',
  status: 'A',
  notificationsActive: true
});

```

Let me know if you may know whats the issue in my environment. Thanks you a lot!!

LucasBrazi06, 2022-04-14:
The password is never stored in clear in the DB lol, only its hash.
Request a password in the front-end and you will receive an email to enter a new password ;-).

AstralisSomnium, 2022-04-21:
AWESOME, now I could register and login*! MANY, MANY thanks Lucas!!

Now, in the dashboard, it does not show me the charging station:
+ I have added in following path `server/dist/assets/charging-station-templates/charging-stations.json` following unzipped file 
[charging-stations.zip](https://github.com/sap-labs-france/ev-server/files/8535616/charging-stations.zip)
  + @LucasBrazi06 I cannot find any error, guidelines nor wiki about the charging station json. What changes do I need to make in my son or what next steps are missing that I can connect my charging station?

[image]

_*PS: The email was not sent, probably some problems with my yahoo smtp settings but I updated in the MongoDB status to "A" and then the login worked._

LucasBrazi06, 2022-04-22:
For the template you can find the Json schema to find out the format.
You are logged into the master tenant, you have to create an organisation an jump into it. 
There, you will find the charging station menu and the onboarding tab to create a connection link.