sap-labs-france/ev-server #3483: Rest Api Problems

mytrkr0, 2022-04-20:
Hello everyone! I tried some function on ev-server but i receive some error like below. it is seems about schema but I don't now exactly. 

What is the problem?

Action: Delete Station
Error: "Can't resolve reference chargingstation#/definitions/id from id chargingstation-delete"

[image]

LucasBrazi06, 2022-04-20:
I don't have any error like this one in master-qa.
Check the schema files (src/assets/schemas/chargingstation/chargingstation.json)

mytrkr0, 2022-04-21:
Thank you very much.

Actually I looked after you said my schema i used wrong schema but i update with master-qa json and it worked.

closed 2022-04-21 by mytrkr0