sap-labs-france/ev-server #3499: Missing porperty 'JsonEndpoint.targetPort' in config.json

Gregor193, 2022-04-27:
Hello 
When starting the server I get the following error message: 

[image]

Attached is my config.json

[config.json.txt](https://github.com/sap-labs-france/ev-server/files/8569622/config.json.txt)

I have set the port in config. Why do I get an error message for this? Have I made a beginner's mistake here?

mytrkr0, 2022-04-27:
Hello,

You can add to config.json

```json
"JsonEndpoint": {
    "baseSecureUrl": "ws://192.168.0.10:8010"
  }
```

instead change it as below

```json
"JsonEndpoint": {
    "baseSecureUrl": "ws://192.168.0.10:8010",
    "targetPort" : "<YOUR_TARGET_PORT>"
  },
```
like this

```json
"JsonEndpoint": {
    "baseSecureUrl": "ws://192.168.0.10:8010",
    "targetPort" : 8010
  },
```

I wish conveniences

Gregor193, 2022-04-27:
Thank you, it worked 
Do you also have any advice regarding the warning monitoring? 
Thanks in advance

LucasBrazi06, 2022-04-27:
> Thank you, it worked Do you also have any advice regarding the warning monitoring? Thanks in advance

It is used in production when you have Prometheus as a monitoring tools, then not mandatory.

closed 2022-05-09 by LucasBrazi06