sap-labs-france/ev-server #3506: TRANSACTIONS - Check Consumption Consistency

ClaudeROSSI, 2022-05-03:
To be on the safe side and avoid billing inconsistent data we need to make sure the meter values  we receive are consistent. 

This does not occur so often. So the first approach will be to just detect and ignore (or cap) inconsistent values and log an erreur.
In the future, if this happens too often we will mark the session as being "IN error" and send a mail to the admins.

So far, the pb is only visible on a particular charging station:

[image]

ClaudeROSSI, 2022-05-03:
@OliveGerste : Let me know if you can take this one.
Thanks,
Claude

LucasBrazi06, 2022-05-09:
@ClaudeROSSI Maybe another developper like Mahdi would be better.

mytrkr0, 2022-06-21:
What is the reason of this Mr. Claude Could you explain it? @ClaudeROSSI

ClaudeROSSI, 2022-06-21:
> What is the reason of this Mr. Claude Could you explain it? @ClaudeROSSI

We faced the situation where a charging station sends inconsistent meter values. We need to detect these inconsistent values and get rid of them.

mytrkr0, 2022-06-21:
Yes I got it. But I mean is what is the reason? Could you find it?