sap-labs-france/ev-server #3533: How can get user id through rest API (is there API service)

mytrkr0, 2022-05-19:
Hello Mr. Serge i looked RestApi services with postman but i couldn't find how can I get user id.

LucasBrazi06, 2022-05-19:
I don't get what you meant :-)
But you can use Swagger: http://localhost:8002/v1/docs/
8002 is the port of your REST server.

closed 2022-05-19 by LucasBrazi06

mytrkr0, 2022-05-20:
Thank you for your answer Mr. Serge. 

Actually i mean was; i can get "User information" from "**Rest Server**" by use "**user ID**".

But when I login to server with **Rest**; there isn't any "**User ID**" related with my **e-mail address** in response, only there is **token**.

In this situation i can't learn information of login person. So should be there "**user ID too**" with "**token**" in Login Successful response.

I hope I was clear :)

LucasBrazi06, 2022-05-20:
The user ID is in the token.
You can decode it.

mytrkr0, 2022-05-20:
Glad to hear that Mr Serge :)

How can i find this decode formula?

mytrkr0, 2022-05-20:
JWT,  okay i found it

Thank you for all Mr. Serge

LucasBrazi06, 2022-05-20:
With that? ;-)
https://jwt.io/

mytrkr0, 2022-05-20:
Definitely yes :)

Thank you repeatedly.