sap-labs-france/ev-server #3548: New User with phone information with REST API

mytrkr0, 2022-05-23:
Hello Mr. Serge

How can i put phone number information in rest request when register new user.

Like this: 

```json
{
    "email": "<E-MAIL>",
    "password": "<PASSWORD>",
    "mobile": "<NUMBER>",
    "acceptEula": true,
    "tenant": "<TENANT>"
}
```

How can i learn it?

LucasBrazi06, 2022-05-23:
In REST endpoint:
http://localhost:8002/v1/docs/#/Authentication/post_auth_signon

mytrkr0, 2022-05-23:
Could you explain you answer Mr. Serge.

LucasBrazi06, 2022-05-23:
It seems that you want to register a user with a mobile phone, I simply pointed out the URL in swagger to do so.

mytrkr0, 2022-05-23:
Actually, what I want to ask is when creating a register by basic mode; transfer phone information.

Registration is created with mail reference. 

Only I don't want update telephone information after registration.

mytrkr0, 2022-05-24:
Hi Mr.Serge i tried to this api url: `{{base_url}}/v1/docs/#/Authentication/post_auth_signon` on postman

with 

```json
{
  "name": "<Name>",
  "firstName": "<Surname>",
  "email": "<Email>",
  "captcha": "<captcha>",
  "password": "<Password>",
  "acceptEula": true,
  "tenant": {{tenant}}
}
```

bot i took this response:

```html
<!DOCTYPE html>
<html lang="en">

<head>
	<meta charset="utf-8">
	<title>Error</title>
</head>

<body>
	<pre>Cannot POST /v1/docs/</pre>
</body>

</html>
```

Is my request correct?

----
How can i fill body for this `{{base_url}}/v1/docs/#/Authentication/post_auth_signon`

Could you give an example?

mytrkr0, 2022-05-30:
Hello Mr. @LucasBrazi06 

`> In REST endpoint: http://localhost:8002/v1/docs/#/Authentication/post_auth_signon
`

I Tried this url but i encountered this error on POSTMAN

`{{base_url}}/v1/docs/#/Authentication/post_auth_signon` --> my url

```html
<!DOCTYPE html>
<html lang="en">

<head>
	<meta charset="utf-8">
	<title>Error</title>
</head>

<body>
	<pre>Cannot POST /v1/docs/</pre>
</body>

</html>
```

Could you give an example with this request or could you explain this request method.

LucasBrazi06, 2022-06-01:
You can debug the code or check/provide the logs.