sap-labs-france/ev-server #3558: Getting siteAreas with parameter "WithChargingStations=true" doesn't work anymore

vladajanjanin, 2022-05-25:
SiteAreas don't contain lists of ChargingStation objects anymore, and the paremeter WithChargingStations seems to be there still.
This is the API call I've been successfully using for a year basically.
[image]
Thanks

LucasBrazi06, 2022-05-25:
I fixed the issue, you should have the charging station IDs in an array tomorrow (after tonight's deployment)

closed 2022-05-25 by LucasBrazi06

vladajanjanin, 2022-05-25:
> I fixed the issue, you should have the charging station IDs in an array tomorrow (after tonight's deployment)

Hey Serge, thanks for a very fast response.
The thing is - previously we could fetch all of the Charging Stations (complete objects) in a single call while fetching all existing Site Areas and it was extremely convenient. 
Being able to see ChargingStation IDs wouldn't provide much since we can fetch ChargingStations for a certain SiteArea ID already, but making an API call for each SiteArea isn't that great compared to how it was earlier. 
If it was an intentional change - ok, we can obviously work around it, but being able to fetch a whole SiteArea in every single charging station object while fetching charging stations, but not the other way around doesn't seem logical.
Thanks once again

vladajanjanin, 2022-05-25:
@LucasBrazi06 
Thanks for the fast reply once again.
It's fine, at least I know that it's working as intended and I should work around it :)
Why I'm asking for all this if you're interested - I'm creating a map filter on our mobile app, and I want to exclude certain SiteAreas from showing based on the connector type of the ChargingStations in that SiteArea. 
It was extremely convenient to have all the information after a single API call instead of fetching all the SiteAreas and then fetching ChargingStations for each SiteArea separately.

LucasBrazi06, 2022-05-25:
@vladajanjanin 
Thanks for the explanation of your use case !
But this is a specific dev at server side to take into account the connector type and filter out the sites/site areas accordingly.
It's a better solution that retrieving all the data and doing the filter in the front-end.
If you have your mobile app then you should also have your own server and you can implement this feature very easily ;-).