sap-labs-france/ev-server #3568: Error: The Tenant ID 'default' is invalid!

Mel0872, 2022-05-29:
Hello,

I managed to set up the ev-server and ev-dashboard (on ubuntu 2004). After I started the server I found a default collection in the database. I could not found the empty-db.zip or empty-db-service.zip for initial setup of the database.
So I only added a SuperAdmin and an Admin user (with the mongo shell). 
As Admin user I logged in and added a onboarding charging station.
The OCPP url looks like: /OCPP16/default/62928f301c595c4e2614bf1c (I got it from the copy to clipboard button).

When I configure my Keba P30 wallbox with the following parameters 

> Chargepoint Identity: 62928f301c595c4e2614bf1c
> OCPP Communication Type: OCPP 1.6 JSON
> Central System Hostname: <SERVER-IP>
> Central System Port: 8010
> Central System Path: OCPP16/default/62928f301c595c4e2614bf1c
> Secure Central System Connection: OFF
> Central System Authentication Method: None
> Predefined Token: <empty>

I get the following erorrs on the log page:
> CentralServer WsServerConnectionOpen  - Open > WS Connection ID 'gAWJp' has been rejected and closed by server due to an exception: The Tenant ID 'default' is invalid!
> CentralServer WsServerConnectionClose - Close > WS Connection ID 'gAWJp' closed by charging station with code '1006', reason: 'Open > WS Connection ID 'gAWJp' has been rejected and closed by server due to an exception: The Tenant ID 'default' is invalid!'
> CentralServer WsServerConnectionOpen  - 62928f301c595c4e2614bf1c The Tenant ID 'default' is invalid! 

I also tried to add a new user with the dashboard and onboard the charging station with this user but I got the same result.

Why could the TenantId default invalid?
Thanks in advance.

LucasBrazi06, 2022-06-01:
You cannot connect charging station in the Master Tenant (ID=default)
You must create a tenant and connect your charging station to it.

closed 2022-06-01 by LucasBrazi06

Mel0872, 2022-10-06:
How can I connect the charging station to a tenant.
I can not find any posibility to do that in the dashboard.

LucasBrazi06, 2022-10-06:
1000€