sap-labs-france/ev-server #3580: Login URL showing 404 error

dignitech-rakesh, 2022-06-03:
I have configured both ev-dashboard and ev-server on my server, and it seems to be accessible from the URL. However, when try to log in from the dashboard panel, it shows 404 error in the console.
[image]

I checked the pm2 log also, and this is what I found:
[image]

closed 2022-06-03 by LucasBrazi06

xEmanuelx, 2022-06-03:
I have the same problem.
@LucasBrazi06 Could you give us support ?

LucasBrazi06, 2022-06-03:
> I have the same problem.
> @LucasBrazi06 Could you give us support ?

Have you checked my reply?