sap-labs-france/ev-server #3787: Billing problem

aliuysal60, 2022-08-14:
Hello;
The normally running billing service has stopped working. Is there any information on this subject? Should I look on my side?

The charging process starts and finishes successfully, I see the price graph increase in the statistics, but the invoices do not appear in the and there is no movement on the Stripe side.

I started having this problem with version 2.6.8

I can't see any records in the logs. Neither on the tenant's and admin side

closed 2022-08-22 by aliuysal60

Zakichanu, 2022-11-05:
HI, do you have any advice how to work this out, I have the same issue...

aliuysal60, 2022-11-05:
No, I couldn't find a solution.
Maybe @LucasBrazi06 can help.

LucasBrazi06, 2022-11-06:
The current version is 2.7.3, do you have the same issue?

aliuysal60, 2022-11-06:
I see the latest version 2.6.10

Zakichanu, 2022-11-06:
> The current version is 2.7.3, do you have the same issue?

In my case, i'm in the 2.7.3 version

LucasBrazi06, 2022-11-06:
The only way to find out what's wrong is the logs!

Zakichanu, 2022-11-06:
I look through the logs and get this : 
[image]

I tried the payment on a charging station from the ev-simulator, I search what it can be, I didn't find anything relevant

Thanks in advance

Zakichanu, 2022-11-08:
A little update, my customer is now accepted and I don't have any issue with it from now on.

So I started to check everything about my stripe account, everything is good about it.

My admin account of the test organisation have credit card affiliate to his account : 
[image]

I then, checked the session and on the tab of refunding, we can see here, we got all my transcations in status "Not Submitted", in stripe the logs are not calling the endpoint to retrieve payment...
[image]

Any hint to solve this? @LucasBrazi06 @ClaudeROSSI

ClaudeROSSI, 2022-11-08:
> I look through the logs and get the same error on two different actions :
> 
> * BillingPaymentMethods
> 
> [image]
> 
> * BillingSynchronizeUser
> 
> [image]
> 
> I tried the payment on a charging station from the ev-simulator, I search what it can be, I didn't find anything relevant
> 
> Thanks in advance

The screenshot shows that the customerID is inconsistent. This means that someone (a STRIPE administrator) went to the stripe account and deleted the customer (or someone change the settings to reference a different STRIPE account). The consequence is that the user on the emobility side is orphan and references something which does not exist anymore on the stripe side.

You should normally never get to that situation. The workaround however is to navigate to the user list and force the synchronization with the billing system.

[image]

This will create a new customer in STRIPE and set again the customerID to a valid value.
However, the former user history (if any) is lost.

Hope this helps,
Claude

ClaudeROSSI, 2022-11-08:
> A little update, my customer is now accepted and I don't have any issue with it from now on.
>
> So I started to check everything about my stripe account, everything is good about it.
>
> My admin account [...]

I'm afraid, I do not understand the question here.
There is no link between STRIPE and the refunding capabilities that Open e-Mobility offers.
These are two different layers!

Let me know what you think!
Regards
Claude

Zakichanu, 2022-11-09:
> The screenshot shows that the customerID is inconsistent. This means that someone (a STRIPE administrator) went to the stripe account and deleted the customer (or someone change the settings to [...]

Hi @ClaudeROSSI,

Thanks for this response, I made this process and it worked, I have to test it out now on a session.

> I'm afraid, I do not understand the question here. There is no link between STRIPE and the refunding capabilities that Open e-Mobility offers. These are two different layers!
>
> Let me know what you [...]

To respond to this, I just realize the misunderstanding I made, thanks to you for this reply!

I will update you if there any other issue.

Zakichanu, 2022-11-14:
Hi again,

I was wondering if the transaction are working even if it is a charging station from the simulator?

Thanks in advance

ClaudeROSSI, 2022-11-15:
> Hi again,
> 
> I was wondering if the transactions are working even if it is a charging station from the simulator?
> 
> Thanks in advance

Transactions work properly as soon as they receive consistent OCPP meter values (with consistent data). The simulator is supposed to mimic the behaviour of regular charging stations. Right? 
When it does its job properly, there is no way server-side to know whether the OCPP calls are coming from a real charger or a fake one. You could even trigger the pricing and billing layers if the corresponding modules are properly configured.

Claude

Zakichanu, 2022-11-18:
Hi! 

Ok, thanks for your response, I will try with a simulator ASAP

Zakichanu, 2022-11-19:
Hi again @ClaudeROSSI,

I tried with ev-simulator. I ran a session on a charging station with a price of 2€/session as you can see here (the data are biased but I don't think it is the most important thing here) : 
[image]

But there is nothing happening on my stripe dashboard...
[image]

I don't see where the problem can be, set apart, maybe, the fact that I am using a simulator instead of a real session with a real car.

Zakichanu, 2022-11-27:
Hi again,

I am still dealing with this issue... If there is any solution (or not), hit me up!