sap-labs-france/ev-server #3792: Remote start not working

KinzaBS, 2022-08-18:


KinzaBS, 2022-08-18:
A customer is complaining :

- Car is charging and you remote stop -> status in finishing
- when you need to start transaction again, button remote start is missing (it has been there before)
[image]

Ramzay, 2022-08-18:
Hello @KinzaBS, the start charging button is only available / allowed  (make sense) when the status is either:
- Available
- Preparing
- Reserved

The Finishing status is temporary, the status that comes afterwards should be 'Available' (which allows a start charge)

Let me know :)

ClaudeROSSI, 2022-08-31:
We had a concrete use-case where restarting the charge while being in status FINISHING makes sense.
So the action is now enabled again.

closed 2022-08-31 by ClaudeROSSI