sap-labs-france/ev-server #3954: Errors with getConfiguration with charging stations

Zakichanu, 2022-11-01:
Hi,

I am currently working with your app and I was trying to add a charging station on a test organization which has 4 connectors.

The bootNotification works well, but, the OcppGetConfiguration is never done and when I try to change configuration, we got this error : 
[image]

Are you familiar with this?

Zakichanu, 2022-11-02:
The thing is, I want to get the logs of the communication between central server and the charging station, how can i see it?

closed 2022-11-02 by Zakichanu

Justin9302, 2023-10-03:
because the hardware(CP) is not fully comply with the OCPP protocol, which should reply with the message like this: getconfigurationkeys:{knwon keys},{unknown keys}.