sap-labs-france/ev-server #3957: Version 2.7.4 - November

ClaudeROSSI, 2022-11-02:


ClaudeROSSI, 2022-11-02:
Version 2.7.4

Only minor changes and some fixes

- 'Security - Libraries have been updated to address security issues and avoid vulnerabilities',
- 'Billing - STRIPE - Code cleanup to avoid using deprecated methods',
- 'Billing - improvement - Some code reorganization for a better abstraction of the underlying implementation',
- 'Authorizations - fix - Assignment of site admin and site owner roles',
- 'Authorizations - fix - Admin can refund charging sessions when being site owner',
- 'Logging - improvement - minor change - application errors logged in their tenant'

changesbyjames, 2022-11-03:
Hey Claude, do you know if this release would include the additional statistics aggregations changes that were merged a couple weeks ago?

closed 2022-11-18 by ClaudeROSSI