sap-labs-france/ev-server #4051: Email Not Arrive to gmail

An-nasMs1817120893, 2023-01-15:
Hi, this might sound stupid, but I'm not able to get the verification email after register my account. 

i have registered my account and it was successful , you can see the mongdb log as shown below 

[image]

However, after that it said that I need to wait for admin to verify my account and check my email however I did not receive any email

below is the log output from mongo db if I tried to log in without verified my account ( I mean how can I verify my account if the email is not arriving in my gmail right ? )

[image]

can someone help me pls , I have spent countless hours trying to solve this , I really need some help :( 

if possible can someone share their config.json file snippets so I can follow pls :( 

I have read the issues which are completely similar to mine however the issue was closed and no direct solution was provided , pls help me :(

thank you.

Zakichanu, 2023-01-15:
Hi,

You need to configure an SMTP server to allow your app to send emails. For your case, if you want to allow the user the authentication in a different way, you can just go to the the users table and change status `P` (for pending) to `A` (active)

Steps in detail : 
- Go to the users section of the default or the organization you want to log in : 
[image]

- Select the user want to activate access
- Then change the `status` attribute to `A` like follow
[image]

Hope this helps :)

An-nasMs1817120893, 2023-01-15:
WOW !! THANK YOU SO MUCH T_T !! this solved my issues !! :)

Zakichanu, 2023-01-15:
Amazing! You can close this issue then

Have a nice day :D

closed 2023-01-15 by An-nasMs1817120893