sap-labs-france/ev-server #4066: EV Simulator - Chargers connecting to Backend

testocpp, 2023-02-01:
Hi

Been working on connection Simulation code to EV server, simulated chargers do send heartbeat message but connection is not established and chargers don't show up on dashboard. Anyone who ran into same issue , looking for some help.

Thanks

testocpp, 2023-02-22:
Apologies for late reply. Yes I am using latest version.

Anmirazik, 2023-02-22:
rename `src/assets/config-template.json` to `src/assets/config.json` in the ev-simulator folder 

After that , you must generate and copy the token from the dashboard as shown below 

[image]

once you have copied the token you should paste it in the supervision URL located in the `src/assets/config.json`

testocpp, 2023-02-22:
Thats what we did. Simulated Chargers do try to connect to backend as per logswe see but they never show up on the dashboard/main screen.

Anmirazik, 2023-02-22:
can you show us the logs from the dashboard which stated that the charger are connected ?