sap-labs-france/ev-server #4201: HTTPS - SSL confirmation

AndyChriss123, 2023-06-03:
test

AndyChriss123, 2023-06-08:
I have managed with nginx as a reverse proxy with SSL enabled and configured. I understand that HTTPS is not a priority for dev environment from other such posts. If anybody needs help with this I'm willing to help out, just let me know.
All the best!

closed 2023-06-08 by AndyChriss123

samarasdelta, 2023-07-25:
Hello @AndyChriss123,
I seem to have the same problem. Could you describe the resolution steps for this particular issue?
Thanks in advance!

AndyChriss123, 2023-07-25:
Hello,
On the dashboard side you can rewrite the startup script for start-ssl (or run-ssl i don't remember exactly) to include your certificate paths. For communication between central server and dashboard or central server and mobile I use nginx as a reverse proxy through which I encrypt the data using the same ssl certificates while maintaining http communication from the config files. I can upload the config files for server dashboard and nginx if you would like but It might be as late as tomorrow.
They have stated somewhere on github in an issue chat that ssl is not a high priority for them as it does not affect dev or something like that so this is only a solution that I came up with while facing this problem.
Good luck and have a great day!