sap-labs-france/ev-server #4209: Cant make the ev server docker run!

UBT-FJ, 2023-07-04:
Constantly getting this!

`2023-07-04 21:35:54 --------------------------------------------------------
2023-07-04 21:35:54  docker-compose-wait 2.7.3
2023-07-04 21:35:54 ---------------------------
2023-07-04 21:35:54 Starting with configuration:
2023-07-04 21:35:54  - Hosts to be waiting for: [ev_mongo:27017]
2023-07-04 21:35:54  - Timeout before failure: 90 seconds 
2023-07-04 21:35:54  - TCP connection timeout before retry: 5 seconds 
2023-07-04 21:35:54  - Sleeping time before checking for hosts availability: 0 seconds
2023-07-04 21:35:54  - Sleeping time once all hosts are available: 3 seconds
2023-07-04 21:35:54  - Sleeping time between retries: 1 seconds
2023-07-04 21:35:54 --------------------------------------------------------
2023-07-04 21:35:54 Checking availability of ev_mongo:27017
2023-07-04 21:35:54 Host ev_mongo:27017 is now available!
2023-07-04 21:35:54 --------------------------------------------------------
2023-07-04 21:35:54 Waiting 3 seconds after hosts availability
2023-07-04 21:35:54 --------------------------------------------------------
2023-07-04 21:35:57 docker-compose-wait - Everything's fine, the application can now start!
2023-07-04 21:35:57 --------------------------------------------------------
2023-07-04 21:35:58 2023-07-04T19:35:58: PM2 log: Launching in no daemon mode
2023-07-04 21:35:58 2023-07-04T19:35:58: PM2 log: App [start:0] starting in -fork mode-
2023-07-04 21:35:58 2023-07-04T19:35:58: PM2 log: App [start:0] online
2023-07-04 21:36:01 7/4/2023, 7:36:01 PM - e-Mobility Server is starting...
2023-07-04 21:36:01 7/4/2023, 7:36:01 PM - NodeJS is started in 'production' mode
2023-07-04 21:36:01 7/4/2023, 7:36:01 PM - Connecting to the Database...
2023-07-04 21:36:01 7/4/2023, 7:36:01 PM - Connecting to 'mongodb'...
2023-07-04 21:36:00 [Warning] Missing property 'Firebase' in config.json
2023-07-04 21:36:01 [Warning] Missing property 'OICPService' in config.json
2023-07-04 21:36:01 [Warning] Missing property 'Monitoring' in config.json
2023-07-04 21:36:01 [Warning] Missing property 'Cache' in config.json
2023-07-04 21:36:01 7/4/2023, 7:36:01 PM - TypeError: Cannot read properties of undefined (reading 'toString')`

UBT-FJ, 2023-07-05:
I have just tried also the node version to install directly to Ubuntu and get the same error

`7/5/2023, 12:25:39 PM - e-Mobility Server is starting...
7/5/2023, 12:25:39 PM - NodeJS is started in 'production' mode
[Warning] Missing property 'Monitoring' in config.json
[Warning] Missing property 'Cache' in config.json
7/5/2023, 12:25:39 PM - Connecting to the Database...
7/5/2023, 12:25:39 PM - Connecting to 'mongodb'...
`7/5/2023, 12:25:39 PM - TypeError: Cannot read properties of undefined (reading 'toString')``

I guess this is a problem and cant seem to find where the problem is:
`7/5/2023, 12:25:39 PM - TypeError: Cannot read properties of undefined (reading 'toString')`

UBT-FJ, 2023-07-05:
It seems that the connection to the db was not successful, with the default settings from the config.json, I had to use the URI  string instead to be able to connect to the db, now it is working, but many all the integration settings are off! is there a way to have these without SAP?

YasineDev, 2023-12-08:
how i can get the uri of the mongodb