sap-labs-france/ev-server #4218: Invoice Unit price!

UBT-FJ, 2023-10-06:


UBT-FJ, 2023-12-06:
anybody?
I also tried in stripe but cant get it to work!

Zakichanu, 2023-12-06:
Hi, can i see your pricing definition?

UBT-FJ, 2023-12-06:
[image]
[image]

Zakichanu, 2023-12-06:
I think the unit price is associated to the flat fee. If you add a flat fee of test, the unit price might return the flat fee

UBT-FJ, 2023-12-06:
but I want to use the Enrgy, isnt that possible?

Zakichanu, 2023-12-06:
What I am telling you is, the first row of the stripe invoice is exclusively for the flat fee on your Price dimension. 

If you want to have the energy price, you can change the code to put it in the first line

UBT-FJ, 2023-12-06:
> What I am telling you is, the first row of the stripe invoice is exclusively for the flat fee on your Price dimension.
> 
> If you want to have the energy price, you can change the code to put it in the first line

Do you happen to know where should I change it? :(

Zakichanu, 2023-12-14:
Did you figured it out, or are you confronting the same issue?

UBT-FJ, 2023-12-14:
> Did you figured it out, or are you confronting the same issue?

Hello @Zakichanu , thanks for getting back to this thread.

I read and followed a lot of code to found out that the `unitPrice` doesnt get with the other values in one of the functions!

Instead I ended up getting that value from the Transactions class.

In `StripeBillingIntegration.ts`, in this function `buildLineItemDescription4PricedDimension()`
there is also this `const unitPrice = Utils.createDecimal(pricedData.unitPrice).toNumber().toLocaleString(userLocale);` which returns 0!

So I had to make it like this `const unitPrice = transaction.pricingModel.pricingDefinitions[0].dimensions.energy.pricedData.unitPrice;` to get something like what you can see in the screenshot.

If you know a better way please let me know.

[image]

Zakichanu, 2023-12-15:
Hi @UBT-FJ,

That's a great way to modify this in my opinion. I never use this part in my case so I don't know if there is a better way for this.

I think we can close this issue then