sap-labs-france/ev-server #4230: Any advice on how to implement Smart Charging?

Zakichanu, 2024-04-19:
I am currently experiencing Smart Charging for this backend, and I would like to know how to make everything working well.

Thanks in advance.

UBT-FJ, 2024-05-02:
did you find a solution?

I'm also interested in this.

Zakichanu, 2024-05-09:
Hi,

I find a way to send charging profile to a CS but didn't find yet a way to give a load balancing for a complete site.

When I find more about that, I will close the issue with a doc of how we manage all these features.

UBT-FJ, 2024-05-30:
hello @Zakichanu 

any updates?

nawikre1, 2024-06-11:
I would already like to know more....i am stil at the beginning of this...

UBT-FJ, 2024-06-11:
I managed to get the server connected with basic authentication, but cant seem to get any charger enable the smart charging!

gasparuff, 2024-08-08:
@UBT-FJ how did you manage to connect with basic auth? I'm struggling with finding the right username and password to connect to the smart charging platform :(. How do you set it up?

UBT-FJ, 2024-08-08:
@gasparuff you need to know a little java, but the main file is here:
“src/main/java/com/sap/charging/server/security/CustomWebSecurityConfigurerAdapter.java”