sap-labs-france/ev-server #4232: docker installation: central server?

jovrstra, 2024-04-24:
Hello,
I'm trying to install the server and dashboard using the docker files (make without submodules). I do not seem to get error messages, all containers are up and running (maildev shows as unhealthy though)... However, when I try to log in, the dashboard gives an error "cannot reach to the central server".

I don't know how to try to reach the central server outside of the dashboard...

Can anyone help? Thanks!

Jörg

PS: docker ps output:
`CONTAINER ID   IMAGE             COMMAND                  CREATED          STATUS                      PORTS                                                                                                                                                                                                                                                      NAMES
1d1c82b3feae   evse-dashboard    "/docker-entrypoint.…"   22 minutes ago   Up 22 minutes               0.0.0.0:3080->80/tcp, :::3080->80/tcp                                                                                                                                                                                                                      evse-dashboard-1
ae76e4d65a3e   evse-server       "docker-entrypoint.s…"   26 minutes ago   Up 26 minutes               0.0.0.0:81->81/tcp, :::81->81/tcp, 0.0.0.0:8000->8000/tcp, :::8000->8000/tcp, 0.0.0.0:8010->8010/tcp, :::8010->8010/tcp, 0.0.0.0:8080->8080/tcp, :::8080->8080/tcp, 0.0.0.0:9090->9090/tcp, :::9090->9090/tcp, 0.0.0.0:9292->9292/tcp, :::9292->9292/tcp   evse-server-1
772724829c7a   mongo:4.2         "docker-entrypoint.s…"   26 minutes ago   Up 26 minutes               0.0.0.0:27017->27017/tcp, :::27017->27017/tcp                                                                                                                                                                                                              evse-mongodb-1
9b233ae7ad08   maildev/maildev   "bin/maildev bin/mai…"   26 minutes ago   Up 26 minutes (unhealthy)   0.0.0.0:1025->1025/tcp, :::1025->1025/tcp, 0.0.0.0:1080->1080/tcp, :::1080->1080/tcp                                                                                                                                                                       evse-mailserver-1
9d4338573e3a   mongo-express     "/sbin/tini -- /dock…"   26 minutes ago   Up 26 seconds               0.0.0.0:8091->8081/tcp, :::8091->8081/tcp                                                                                                                                                                                                                  evse-mongo-express-1`

UBT-FJ, 2024-05-02:
it is not enough to see the view of the docker here, can you post the config of both dashboard and server (if you didnt fix it so far)

jovrstra, 2024-05-02:
I'm going over it with our system administrator. He already noticed that the mongodb dockerfile still contains link, which has been made obsolete. We'll check further, but at the moment I also have to wait for his input. I'll post back anything that can help others.

Anasonic, 2024-06-19:
Hallo @jovrstra  did you solve the problem i have the same problem

jovrstra, 2024-06-19:
Unfortunately not. I If I remember well the main problems we had with the configuration scripts of MongoDB, but it started to take up too much time and we started looking for another solution. It is a shame, as this project looked perfect for us.
Sorry I cannot be of help...

UBT-FJ, 2024-06-19:
For the connection it is very easy to make it!

Actualy best is to use the string, but before that you need to setup the DB, like users, create database, activate replica etc.

In the docker folder, there is a script that does all of this automatically.

Can you tell us what went wrong?

jovrstra, 2024-06-19:
I don't remember and no longer have the setup to verify...  But maybe Anasonic can provide feedback.

Anasonic, 2024-06-28:
I tried but 😔 Can you help us with a paid fees @UBT-FJ  
Eng.anas00@gmail.com 
Ws:00967738306415