sap-labs-france/ev-server #4234: Hubject OICP connection

UBT-FJ, 2024-05-10:
Hello,

anybody did this?

There are the endpoints in the code but don't know how to connect to them!

Is it possible to connect only through the CPO Identification? which it requires only the:
- Country Code
- Party ID (I guess this is the provider/evse operator ID)
- Private key (do we copy here the code from the file that was generated?)
- Certificate (do we copy the PEM signed code that we got from them?)

UBT-FJ, 2024-05-27:
Hello,

I came back since I think I need a little more help about this!

I found a small issue about the create Roaming Connections and posted here:
`https://github.com/sap-labs-france/ev-dashboard/issues/3553#issuecomment-2124615873`

But still facing the issue about the Service URL, which is required by Hubject to send remote start/stop.

I found some code in the server, and added the config to start the OICP server but cant seem to find the proper way on how to do this.

UBT-FJ, 2024-05-28:
It is really strange that none of the contributors are ever replying to any of the issues! "at leaast mine :("

How can we have the solution setup if not from their help?

UBT-FJ, 2024-05-28:
I'm going to write here my journey for this and hopefully I will also help somebody else!

I managed to get the Service URL, from the CPOService file, this:
`public static readonly PATH = '/:protocol/:role/:version/:tenantSubdomain/api/oicp/:module/:endpointVersion/providers/:providerID/:endpoint/:endpointAction?';`
So I made the actual URL like this:
https://OICPserverdomain/https/cpo/v1/germany (the rest comes directly from Hubject)

And the communication now is ok.

Another problem is that when I try to do remote start I get `Remote Start rejected by Charging Station` and in the server I get this `Cannot read properties of undefined (reading 'length')`
I think that is because of this idTag that OICP uses and I cant add it in the rfid list:
`"idTag": "NOTAGID-OICPREMOTE"`

And also the local users cant start the charging process for the Public chargers! getting the `User does not belong to the local organization` although the user is in that organization. 
If I disable that charger as public that user can start charging!

nawikre1, 2024-06-11:
i am sorry i cannot contribute...i am also searching my a** of for information on this csms....but appertently you have to do everyhting yourself...hahahaha I am now starting my journey with Hubject. I will follow this and you are helping me for sure.

UBT-FJ, 2024-06-11:
Good luck, please share your experience.

eyadosama, 2024-10-27:
@UBT-FJ Hello. Can you please provide me with any way of communicating with you? I have some questions regarding this repository. I really appreciate any help you can provide.