sap-labs-france/ev-server #4235: Manual somewhere??

nawikre1, 2024-06-11:
Hi there,

i am working with this system for about 1 month now. I am searhcing and trying. I have connected a couple of ev-chargers to the csms, i even got stripe connected....hahha but i still have so many questions on even where some options can be set.
like a simple thing. 
1. When i login to the backend part i can add organisations, SEE cars and add users and see logs.
But what is the purpose of the sections "cars". I cannot ADD cars, i can just SYNC cars, but it does nothing....within the past month i have been charging 2 cars 3 x per week....but i still see that standard 3 cars in my list. HOW can i manage this. can i delete these cars, can i add cars.
2. When i login the csms front end i can also go to cars....i can ADD a car, but i MUST select a car from the car catalog that only has 3 strange cars (that are seen in the back end server).
3. I have tried to add RFID cars. I have a rfid reader that reads the UID. When i programmed it into the csms i am HOPING that the charging station communicated with the csms when i swipe my rfid card and gives me permission to charge...but nothing happens. I tried to play with the ocpp code with localauthorisation etc....but i am lacking in knowledge.
4. Can someone explain what the real purpose is for the " assets" section. Can i just enter my assets like my cars, buildings, ev chargers, home batteries etc...but what is the purpose...can i use it within the csms somehow....no idea.
5. in "organisation"  i can enter a company, but also sites and a site areas...how do i define this....what do i enter in sites of site area. is this as a zoom-in...like sites i would enter a city and site area is a more specific area, like west part of the city?
6. In the "integration settings"  in "Billing"  My stripe connection is good, but at the lower bottum i need to fill in VAT. It is empy with me. Do i need to setup these details in stripe and it syns?

and there are many more questions about such issues. I am willing to learn and am learning everyday. But i am a simple guy that ants to set this up so that it can work.

Thanks

Iwan

laptar-ls, 2024-07-09:
Hi @nawikre1, could you please help to deploy this project locally, I'm trying to do it via docker-compose but I can't?

nawikre1, 2024-07-09:
@laptar-ls ..i have never worked with docker, so I could not help you with this. But I know someone that can help you...but it will cost a bit. He is absolutely not expensive. He helped me a lot with setting up the SMTP server etc so I would get notifications etc. Also he helpen with the captcha and google maps key.
If you need his help, send me a message at Iwan (at) ldr-ict(dit)nl